pycom/pymakr-vsc #144: Serialport module error - VSCode

GregorR1, 2021-08-06:
## What are the steps to reproduce this issue?
1.Open VScode with pymakr installed
2. 

## What happens?
Recieve the following error:

There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)

Operating system: Windows 10

VSCode version: 1.59.0

Pymakr version: v1.1.12

ossonts, 2021-08-06:
Same here, last time I rolled back a version of VSCode to get this working, but this time it would be good to see what is going on , any ideas on debug options?

This is a fresh install Win 10 + VSCode + node. Attempt to connect and 
```
There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)
```

Pymakr v1.1.12

VSCode:
Version: 1.59.0 
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
OS: Windows_NT x64 10.0.19043

SubaruArai, 2021-08-07:
I can reproduce it too:
Pymakr v1.1.12

VSCode:
Version: 1.59.0
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:13:20.182Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Linux x64 5.4.135-1-MANJARO

here are the logs from the developer console:
```
workbench.desktop.main.js:62 [Extension Host] (node:14654) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.(Use `code --trace-deprecation ...` to show where the warning was created)
S @ workbench.desktop.main.js:62
```
```
[Extension Host] Error while loading serialport library
```
```
workbench.desktop.main.js:62 [Extension Host] (node:14654) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.(Use `code --trace-deprecation ...` to show where the warning was created)
S @ workbench.desktop.main.js:62
workbench.desktop.main.js:62 [Extension Host] Error while loading serialport library
workbench.desktop.main.js:62 [Extension Host] Error: Could not locate the bindings file. Tried: → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/Debug/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/Release/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/out/Debug/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/Debug/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/out/Release/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/Release/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/default/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/compiled/14.16.0/linux/x64/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node → /home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/lib/binding/node-v89-linux-x64/bindings.node	at bindings (/home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/bindings/bindings.js:126:9)	at Object.<anonymous> (/home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/lib/linux.js:2:36)	at Module.u._compile (/opt/visual-studio-code/resources/app/out/vs/loader.js:4:1315)	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)	at Module.load (internal/modules/cjs/loader.js:982:32)	at Module._load (internal/modules/cjs/loader.js:823:14)	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)	at Function.n._load (/opt/visual-studio-code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:28060)	at Function.P._load (/opt/visual-studio-code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:24648)	at Function.l._load (/opt/visual-studio-code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:59608)	at Module.require (internal/modules/cjs/loader.js:1006:19)	at v (/opt/visual-studio-code/resources/app/out/vs/loader.js:4:698)	at Object.<anonymous> (/home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/lib/index.js:14:22)	at Module.u._compile (/opt/visual-studio-code/resources/app/out/vs/loader.js:4:1315)	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)	at Module.load (internal/modules/cjs/loader.js:982:32)	at Module._load (internal/modules/cjs/loader.js:823:14)	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)	at Function.n._load (/opt/visual-studio-code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:28060)	at Function.P._load (/opt/visual-studio-code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:24648)	at Function.l._load (/opt/visual-studio-code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:59608)	at Module.require (internal/modules/cjs/loader.js:1006:19)	at v (/opt/visual-studio-code/resources/app/out/vs/loader.js:4:698)	at Object.<anonymous> (/home/username/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/serialport/lib/index.js:2:17)	at Module.u._compile (/opt/visual-studio-code/resources/app/out/vs/loader.js:4:1315)	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)	at Module.load (internal/modules/cjs/loader.js:982:32)	at Module._load (internal/modules/cjs/loader.js:823:14)	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)	at Function.n._load (/opt/visual-studio-code/resources/app/out/vs/workbench/services/extensions/node/extensionHostPro [... cut ...]
$onExtensionActivationError @ workbench.desktop.main.js:2038
DevTools failed to load source map: Could not load content for https://ticino.blob.core.windows.net/sourcemaps/379476f0e13988d90fab105c5c19e7abc8b1dea8/core/vs/workbench/workbench.desktop.main.js.map: Load canceled due to load timeout
```
```
DevTools failed to load source map: Could not load content for https://ticino.blob.core.windows.net/sourcemaps/379476f0e13988d90fab105c5c19e7abc8b1dea8/core/vs/workbench/workbench.desktop.main.js.map: Load canceled due to load timeout
```

victorhooi, 2021-08-08:
I am hitting this issue as well - specifically that after installing the Pymakr extension in VSCode, I get the following error message:

> There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)

[image]

Pymakr v1.1.12

VS Code version info:
```
Version: 1.60.0-insider
Commit: 49af1cbe00271ac2ea36cfc5f7352fe31a7b0094
Date: 2021-08-06T06:42:17.124Z (1 day ago)
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Darwin x64 20.6.0
```

In VS Code, if I then go to Help, Toggle Developer Console, I can go to the console tab and I see this:

```
Activating extension 'pycom.pymakr' failed: Could not locate the bindings file. Tried:
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/Release/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/Debug/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/out/Release/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/Release/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/default/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/compiled/14.16.0/darwin/x64/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /Users/victorhooi/.vscode-insiders/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/lib/binding/node-v89-darwin-x64/bindings.node.
$onExtensionActivationError @ workbench.desktop.main.js:sourcemap:1977
```

Bliph, 2021-08-08:
Same issue here:
```
Version: 1.59.0 (user setup)
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:13:12.822Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Windows_NT x64 10.0.19043
```

**Workaround**: Downgrade VSCode to 1.58.0 (but this is a **pycom** issue!)

snebot-bg, 2021-08-09:
Same issue here. Tried downgrading VSCode to 1.58.2 with no luck.

```
Version: 1.58.2 (user setup)
Commit: c3f126316369cd610563c75b1b1725e0679adfb3
Date: 2021-07-14T22:10:15.214Z
Electron: 12.0.13
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Windows_NT x64 10.0.19042
```

catastrophical, 2021-08-09:
Can confirm, having same issue:

[image]

calvfoo, 2021-08-10:
er.. same issue here after updating to VSCode

Version: 1.59.0 (system setup)
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:13:12.822Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Windows_NT x64 10.0.19042

YellowFly2, 2021-08-10:
same issue with mac
[image]

GregorR1, 2021-08-10:
I tried downgrading to:

Version: 1.58.2 (user setup)
Commit: c3f126316369cd610563c75b1b1725e0679adfb3
Date: 2021-07-14T22:10:15.214Z
Electron: 12.0.13
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Windows_NT x64 10.0.19042

This appears to have solved the issue.

ossonts, 2021-08-10:
I downgraded to VSCode 1.56.1 (link here https://code.visualstudio.com/updates/v1_56 ) 

Don't click the download button (that is the latest version) but rather click the link on line 3/4

[image]

This is not a fix, just a work around for us ...  but otherwise all development has halted here. 

This has happened before and it took ages to fix, I am happy to have a look at the issue, but would need some tips on how to best go about debugging this. 

We really need to be on the latest stable VSCode (FYI insiders version also fails)

calvfoo, 2021-08-11:
rollback VS Code to v1.55 and it worked again

Version: 1.55.2 (system setup)
Commit: 3c4e3df9e89829dce27b7b5c24508306b151f30d
Date: 2021-04-13T09:35:57.887Z
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19042

William04A, 2021-08-12:
I confirm having this error on the latest VSCode version both on Ubuntu and on Windows.
Time to downgrade, I guess...

moefear85, 2021-08-13:
Having same issue, but on Ubuntu. I can't believe this keeps happening, despite it being so easy to code an editor, or VS code extension, that auto-syncs with micropython through the standard python serial module (which itself is always working). Once I'm done with my current project, I'll code one myself, and upload it for everyone. Pymakr isn't the endall on this one. It's beyond me why they drag nodejs, electron, telnet & others into this.

phires, 2021-08-16:
Still not working after downgrading to 1.58.2 on Mac.

reimerp, 2021-08-16:
Confirm not working (Ubuntu 20.04 / msc 1.59.0 / pymakr 1.112):

first got _Could not locate the bindings file_, than I tried some symlinking, than _npm install @serialport/bindings_
this generates 
~/.vscode/extensions/pycom.pymakr-1.1.1/node_modules/@serialport/bindings/build/Release/bindings.node
file but msc fails to load:
_Error: Module did not self-register_

Versions:

<details>

Version: 1.59.0
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:13:20.182Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Linux x64 5.4.0-77-generic

Python 3.8.10
</details>

snebot-bg, 2021-08-17:
> Still not working after downgrading to 1.58.2 on Mac.

 @phires Try downgrading to 1.58.0, that seemed to work for me. https://update.code.visualstudio.com/1.58.0/win32-x64-user/stable

ossonts, 2021-08-18:
I am running 1.58.2 no issues 
BUT
If you dont untick the auto update it will update automatically , it does this by default. 

[image]

For those with issues at 1.58 -- check help-> About (version)

[I am still looking for tips to debut and solve this issue, as I need to run on the latest VSCode]

William04A, 2021-08-22:
Is there any update from PyCom about this issue whatsoever?

jakobrosenberg, 2021-09-03:
@William04A we are currently testing a new version, which we hope to release shortly.

Closing in favor of https://github.com/pycom/pymakr-vsc/issues/119 but feel free to continue the conversation.

closed 2021-09-03 by jakobrosenberg

Ladybrinx, 2022-03-31:
I just found this after the March 2022 update, same issue, I was able to rollback to the February 2022 version and it works fine again... whew... talk about heart attack! This version: Update 1.65.1

javiermarasco, 2022-04-01:
I can confirm, not working after an automatic update to March 2022 version, February 2022 works perfectly.

Msprg, 2022-04-01:
Exactly same here - just glad to see it's not my wrongdoing. Guess I'll be rolling back the update as well 🤷‍♂️

jakobrosenberg, 2022-04-02:
Thanks for the feedback everyone. 🙏 We're trying to solve this and will let you know as soon as we have a solution.

Our new [Pymakr 2 preview](https://marketplace.visualstudio.com/items?itemName=pycom.pymakr-preview) uses N-API bindings to solve these issues. If these prove reliable, we may upgrade the original Pymakr to N-API bindings as well.

andreikee, 2022-04-04:
The same problem after the March update!