pycom/pymakr-vsc #145: Pymakr extension fails to install on Atom 1.58.0

victorhooi, 2021-08-08:
I first tried to use Pymakr with VS Code, which is my main editor. However, this failed to install properly, with a serialport error (https://github.com/pycom/pymakr-vsc/issues/144#issuecomment-894720375).

I next did a fresh install of Atom 1.58.0.

I then tried to install the Pymakr extension using the instructions here - https://docs.pycom.io/gettingstarted/software/atom/.

However, the installation fails with the following error message:

```
Installing “pymakr@2.1.13” failed.Hide output…

> @serialport/bindings@8.0.8 install /private/var/folders/j1/2g8b6s4n3v7gc_rxd_zj8j2w00_sn4/T/apm-install-dir-202178-19894-1ea325y.ufsq/node_modules/pymakr/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

npm WARN deprecated crypto@1.0.1: This package is no longer supported. It's now a built-in Node module. If you've depended on crypto, you should switch to the one that's built-in.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated parser-byte-length@1.0.2: renamed to @serialport/parser-byte-length
npm WARN deprecated parser-cctalk@1.0.2: reanmed to @serialport/parser-cctalk
npm WARN deprecated parser-delimiter@1.0.2: reanmed to @serialport/parser-delimiter
npm WARN deprecated parser-readline@1.0.2: reanmed to @serialport/parser-readline
npm WARN deprecated parser-ready@1.0.2: reanmed to @serialport/parser-ready
npm WARN deprecated parser-regex@1.0.2: reanmed to @serialport/parser-regex
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
prebuild-install WARN install No prebuilt binaries found (target=9.4.4 runtime=electron arch=x64 libc= platform=darwin)
No receipt for 'com.apple.pkg.CLTools_Executables' found at '/'.

No receipt for 'com.apple.pkg.DeveloperToolsCLILeo' found at '/'.

No receipt for 'com.apple.pkg.DeveloperToolsCLI' found at '/'.

gyp: No Xcode or CLT version detected!
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:223:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Darwin 20.6.0
gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /private/var/folders/j1/2g8b6s4n3v7gc_rxd_zj8j2w00_sn4/T/apm-install-dir-202178-19894-1ea325y.ufsq/node_modules/pymakr/node_modules/@serialport/bindings
gyp ERR! node -v v12.14.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN notsup Unsupported engine for pymakr@2.1.13: wanted: {"atom":">=1.41.0","node":">=6.3.0 <=7.0.0"} (current: {"node":"12.14.1","npm":"6.14.13"})
npm WARN notsup Not compatible with your version of node/npm: pymakr@2.1.13
npm WARN enoent ENOENT: no such file or directory, open '/private/var/folders/j1/2g8b6s4n3v7gc_rxd_zj8j2w00_sn4/T/apm-install-dir-202178-19894-1ea325y.ufsq/package.json'
npm WARN apm-install-dir-202178-19894-1ea325y.ufsq No description
npm WARN apm-install-dir-202178-19894-1ea325y.ufsq No repository field.
npm WARN apm-install-dir-202178-19894-1ea325y.ufsq No README data
npm WARN apm-install-dir-202178-19894-1ea325y.ufsq No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/victorhooi/.atom/.apm/_logs/2021-08-08T00_26_19_369Z-debug.log
```

It's a little disappointing that Pymakr doesn't work on either VS Code or Atom currently - is there any other platform it does currently work on? Keen to use the board! =)

carlskii, 2021-08-11:
+ 1 - same issue on a Mac. Doesn't work with VSCode either. I don't see the point of these extensions if they are not maintained. Better to pull them completely and state they are deprecated.

AlexanderPilhar, 2021-08-16:
I had the very same problem... But I found a solution that works for me (other than downgrading VS Code zu previous version):

You can [install VS Code in Portable Mode](https://code.visualstudio.com/docs/editor/portable) next to your normal version of VS Code! Visual Studio Code Portable can be [installed in a previous version via chocolatey](https://community.chocolatey.org/packages/vscode.portable/1.58.2) (one which is compatible to current PyMakr).

Note: You may need to disable auto update for Visual Studio Code Portable in settings.
Tip: Use another theme for Visual Studio Code Portable so you can easy see which one you are using.

I hope this helps!

jakobrosenberg, 2021-09-07:
We have released `v1.1.13` which fixes this issue for VSCode. Would love to get some feedback from Atom users to know if this has been fixed for all.