pycom/pymakr-vsc #146: Pymakr does not work under VS Code on Ubuntu 21.04 - command 'pymakr.extra.getSerial' not found)

victorhooi, 2021-08-08:
I am trying to setup VS Code with Pymark under Ubuntu 21.04.

First, I tried VS Code on my primary laptop (macOS). However, this does not appear to work on current builds of VS Code (https://github.com/pycom/pymakr-vsc/issues/144#issuecomment-894720375).

Secondly, I then tried installing the current Atom editor as well - however, this resulted in another error - https://github.com/pycom/pymakr-vsc/issues/145.

Thirdly, I grabbed a second laptop running the latest Ubuntu, and I'm trying on there. The actual Pymakr extension appears to install.

However, nothing happens when I plug in the LoPy4/Expansion board (However, on the board itself, the USB and LED both light up, and the LED on the LoPy4 is slowly flashing blue).

When I go to All Commands -> Pymark -> Extra - List Serial orts, I get this error:

> Command 'Pymakr > Extra's > List serial ports' resulted in an error (command 'pymakr.extra.getSerial' not found).

[image]

If I turn on Developer Tools in VS Code, and open up the Console pane, I get:

```
workbench.desktop.main.js:sourcemap:2038 Activating extension 'pycom.pymakr' failed: Could not locate the bindings file. Tried:
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/Release/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/Debug/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/out/Release/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/Release/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/build/default/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/compiled/14.16.0/linux/x64/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /home/victorhooi/.vscode/extensions/pycom.pymakr-1.1.12/node_modules/@serialport/bindings/lib/binding/node-v89-linux-x64/bindings.node.
```

carlskii, 2021-08-11:
+ 1 - I'm getting this too

MichaelWohlers, 2021-08-14:
I have this issue on Mint 20 also

Antos31, 2021-08-16:
Hi, I have the same issue on Raspbian. I think it is an issue of the extension itself, no command is recognised....

jprodriguez-nbs, 2021-08-17:
Hello,
I have hit this problem both in Ubuntu and Windows. My understanding is that it is related to VS Code 1.59.0 using a higher version of electron than the version used by pymaker serialport module. The problem can be clearly identified with "Help/Toggle Developer Tools" and finding the "'pycom.pymakr' failed: Could not locate the bindings file" trace.

I tried recompiling the serialport module, upgrading node, etc, but I am not an expert in node and did not succeed updating pymakr serialport module to work with the new vscode.

In both operating systems I have "fixed" the problem downgrading VSCode to version 1.58.2

Ubuntu: 
apt install code=1.58.2-1626302803

Note: You need to disable the automatic update of vscode if you want to keep this version until a proper fix is published.

jakobrosenberg, 2021-09-07:
This issue should be permanently solved in `v1.1.13`. You can read more here https://github.com/pycom/pymakr-vsc/issues/119#issuecomment-913712274

Should you still have issues, please let us know. 🙂

advorkovyy, 2021-09-07:
Same issue with `1.1.13` on:
- Windows 
- WSL: Ubuntu 20.04
- MacOS

jakobrosenberg, 2021-09-07:
@advorkovyy thanks for the feedback. Would you be able to share your VSCode logs?

ArnaudPalgen, 2021-09-07:
Same issue with `1.1.13` on Ubuntu 20.04.3

VS code version:
```
Version: 1.60.0
Commit: e7d7e9a9348e6a8cc8c03f877d39cb72e5dfb1ff
Date: 2021-09-01T10:43:02.136Z
Electron: 13.1.8
Chrome: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.11.0-27-generic snap
```
logs:
```
[2021-09-07 15:50:44.554] [exthost] [error] Activating extension pycom.pymakr failed due to an error:
[2021-09-07 15:50:44.554] [exthost] [error] Error: /snap/core/current/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /home/arnaud/.vscode/extensions/pycom.pymakr-1.1.13/node_modules/@serialport/bindings/build/Release/bindings.node)
	at process.func [as dlopen] (electron/js2c/asar_bundle.js:5:1846)
	at Object.Module._extensions..node (internal/modules/cjs/loader.js:1185:18)
	at Object.func [as .node] (electron/js2c/asar_bundle.js:5:1846)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:28190)
	at Function.E._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:24778)
	at Function.o._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:103:59872)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at v (/snap/code/73/usr/share/code/resources/app/out/vs/loader.js:4:698)
	at bindings (/home/arnaud/.vscode/extensions/pycom.pymakr-1.1.13/node_modules/bindings/bindings.js:112:48)
	at Object.<anonymous> (/home/arnaud/.vscode/extensions/pycom.pymakr-1.1.13/node_modules/@serialport/bindings/lib/linux.js:2:36)
	at Module.u._compile (/snap/code/73/usr/share/code/resources/app/out/vs/loader.js:4:1315)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:28190)
	at Function.E._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:24778)
	at Function.o._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:103:59872)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at v (/snap/code/73/usr/share/code/resources/app/out/vs/loader.js:4:698)
	at Object.<anonymous> (/home/arnaud/.vscode/extensions/pycom.pymakr-1.1.13/node_modules/@serialport/bindings/lib/index.js:14:22)
	at Module.u._compile (/snap/code/73/usr/share/code/resources/app/out/vs/loader.js:4:1315)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:28190)
	at Function.E._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:24778)
	at Function.o._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:103:59872)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at v (/snap/code/73/usr/share/code/resources/app/out/vs/loader.js:4:698)
	at Object.<anonymous> (/home/arnaud/.vscode/extensions/pycom.pymakr-1.1.13/node_modules/serialport/lib/index.js:2:17)
	at Module.u._compile (/snap/code/73/usr/share/code/resources/app/out/vs/loader.js:4:1315)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:28190)
	at Function.E._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:24778)
	at Function.o._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:103:59872)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at v (/snap/code/73/usr/share/code/resources/app/out/vs/loader.js:4:698)
	at Object.<anonymous> (/home/arnaud/.vscode/extensions/pycom.pymakr-1.1.13/lib/connections/pyserial.js:20:18)
	at Module.u._compile (/snap/code/73/usr/share/code/resources/app/out/vs/loader.js:4:1315)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.i._load (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:28190)
[... cut ...]
	at Function._callActivate (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:99:14576)
	at /snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:99:12779
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
	at async Promise.all (index 0)
	at async v.$activate (/snap/code/73/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:99:20108)
```

rdagger, 2021-09-07:
Same problem on Windows with VS Code 1.60

```
[2021-09-07 09:34:07.425] [exthost] [error] Activating extension pycom.pymakr failed due to an error:
[2021-09-07 09:34:07.425] [exthost] [error] Error: Command failed: npx electron-rebuild -v 13.1.8
- Searching dependency tree
Attempting to build a module with a space in the path
See https://github.com/nodejs/node-gyp/issues/65#issuecomment-368820565 for reasons why this may not work
× Rebuild Failed
[31m
An unhandled error occurred inside electron-rebuild[39m
[31mnode-gyp failed to rebuild 'c:\Users\Registered User\.vscode\extensions\pycom.pymakr-1.1.13\node_modules\@serialport\bindings'.
For more information, rerun with the DEBUG environment variable set to "electron-rebuild".

Error: Could not find any Visual Studio installation to use

Error: node-gyp failed to rebuild 'c:\Users\Registered User\.vscode\extensions\pycom.pymakr-1.1.13\node_modules\@serialport\bindings'.
For more information, rerun with the DEBUG environment variable set to "electron-rebuild".

Error: Could not find any Visual Studio installation to use

    at NodeGyp.rebuildModule (C:\Users\Registered User\AppData\Local\npm-cache\_npx\c78fbb4f3949be78\node_modules\electron-rebuild\lib\src\module-type\node-gyp.js:109:19)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async ModuleRebuilder.rebuildNodeGypModule (C:\Users\Registered User\AppData\Local\npm-cache\_npx\c78fbb4f3949be78\node_modules\electron-rebuild\lib\src\module-rebuilder.js:94:9)
    at async Rebuilder.rebuildModuleAt (C:\Users\Registered User\AppData\Local\npm-cache\_npx\c78fbb4f3949be78\node_modules\electron-rebuild\lib\src\rebuild.js:226:9)
    at async Rebuilder.rebuild (C:\Users\Registered User\AppData\Local\npm-cache\_npx\c78fbb4f3949be78\node_modules\electron-rebuild\lib\src\rebuild.js:184:17)
    at async C:\Users\Registered User\AppData\Local\npm-cache\_npx\c78fbb4f3949be78\node_modules\electron-rebuild\lib\src\cli.js:154:9[39m

	at checkExecSyncError (child_process.js:625:11)
	at execSync (child_process.js:661:15)
	at electron/js2c/asar_bundle.js:5:12288
	at buildBinaries (c:\Users\Registered User\.vscode\extensions\pycom.pymakr-1.1.13\pymakr.js:6:5)
	at activate (c:\Users\Registered User\.vscode\extensions\pycom.pymakr-1.1.13\pymakr.js:10:5)
	at Function._callActivateOptional (c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:99:14900)
	at Function._callActivate (c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:99:14576)
	at c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:99:12779
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
	at async Promise.all (index 0)
	at async v.$activate (c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:99:20108)
[2021-09-07 09:34:11.946] [exthost] [info] extension host terminating: renderer closed the socket
[2021-09-07 09:34:11.961] [exthost] [error] ProxyResolver#resolveProxy Canceled: Canceled
	at Object.n [as canceled] (c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:5:1157)
	at i._remoteCall (c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:13617)
	at Proxy.N.<computed>.G.charCodeAt.N.<computed> (c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:10012)
	at y.resolveProxy (c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:98:38921)
	at Object.resolveProxy (c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:103:57892)
	at useProxySettings (c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\node_modules.asar\vscode-proxy-agent\out\index.js:159:16)
	at c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\node_modules.asar\vscode-proxy-agent\out\index.js:107:13
	at c:\Users\Registered User\AppData\Local\Programs\Microsoft VS Code\resources\app\node_modules.asar\vscode-proxy-agent\out\index.js:335:13
	at processTicksAndRejections (internal/process/task_queues.js:93:5) 
[2021-09-07 09:34:12.962] [exthost] [info] exiting with code 0
```

jakobrosenberg, 2021-09-09:
For those that have issues with `v.1.1.13`, please see https://github.com/pycom/pymakr-vsc/issues/157#issuecomment-915973179

Tarun75, 2021-09-10:
I too have same problem with VS code on Rpi4, running Raspbian OS.Pls provide solution

VS version
Version: 1.58.0
Commit: 2d23c42a936db1c7b3b06f918cde29561cc47cd6
Date: 2021-07-08T06:53:04.182Z
Electron: 12.0.13
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Linux arm 5.10.17-v7l+

jakobrosenberg, 2021-09-10:
@Tarun75 would you be able to try the 1.1.14 beta?

Tarun75, 2021-09-10:
Already tried that

jakobrosenberg, 2021-09-14:
@Tarun75 sorry about the slow reply. I've been told that we don't officially support ARM, but it's something we're looking into. Until then I might be able to help you build the bindings manually if you're interested.

Antos31, 2021-09-14:
I will be interested into the answer for ARM. I have an Raspberry Pi 4 as
well 😊

sw0rdd, 2021-11-11:
I have the same problem on ubuntu 20.10

"There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)"

jakobrosenberg, 2021-11-11:
@sw0rdd can I ask you to share the output from the develop console? You can open it by clicking `Help` -> `Toggle Developer Tools`.

sw0rdd, 2021-11-11:
[image]

what exactly do you want to see? @jakobrosenberg 
I am a beginner to both linux and vs code

elmasria, 2021-11-11:
Hi @sw0rdd,

Thank you for sharing the details,

can I ask you to send the screenshot again but make sure you select the `console` section;

[image]

Best Regards,
Ahmad El Masri

sw0rdd, 2021-11-14:
@elmasria hi, everything is working fine now, I just downgraded vs code and updated it again and I got pymakr to work, thank you

ShocKwav3, 2022-02-21:
Encountering this error in both ubuntu and mac with latest version of vscode. Any help? Do I need to downgrade?

jprodriguez-nbs, 2022-02-21:
Hello,

Please specify the vscode version that you use so we  can compare.
I am using pymakr with the latest version of vscode in Ubuntu 20.10 without
problems.

BR,
JP

jprodriguez-nbs, 2022-02-22:
Hello,
The versions I use are
vscode: 1.64.2
pymakr: 1.1.17
ubuntu: 21.10
These are working fine. What versions do you use?
BR,
JP