pycom/pymakr-vsc #147: There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)

antifm, 2021-08-17:
Having this error running on Pymakr v1.1.12 on vscode version1.59.

How can I fix this?

Thanks

William04A, 2021-08-17:
@antifm I am not sure when PyCom plans to fix this, because in the meantime the solution is to downgrade Visual Studio Code. See this issue which reports the same thing: https://github.com/pycom/pymakr-vsc/issues/144

AlexanderPilhar, 2021-08-17:
Also, have a look at my solution [installing VS Code Portable in compatible version for PyMakr](https://github.com/pycom/pymakr-vsc/issues/145#issuecomment-899455606)!

antifm, 2021-08-17:
@William04A That sucks. Thanks for the redirect!
 @AlexanderPilhar, thank you for the solution. Got it working through the portable mode! Greatly appreciated!

Cheers to the both of you

jakobrosenberg, 2021-09-03:
Closing in favor of https://github.com/pycom/pymakr-vsc/issues/119

closed 2021-09-03 by jakobrosenberg