pycom/pymakr-vsc #149: error with your serialport module, Pymakr will likely not work properly

PrimeTime416, 2021-08-20:
Hello Pymakr stopped working for me, please note I tried it both under linux and Windows 10 (VM).  If you open VSCode  you will get a message saying:

_There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)_

I have never had an issue with Pymakr until now, please help, thanks all.

tronto, 2021-08-22:
I'm having the same issuse. Running VS code 1.59.1 on mac 10.15.7

schulte2003, 2021-08-22:
I have the same issue with Mac OS 11.5.2, VS Code 1.58.2 on a M1 Chip

anomaly256, 2021-08-24:
Same with vscode 1.59.1 and pymakr extension v1.1.12 on an amd64 debian 10 linux

swordfishBob, 2021-08-24:
Me also, Win 10, VS Code.

PrimeTime416, 2021-08-24:
I am not sure if this is related to the vsCode trust verification? the code looked, looks like it use node.js serialport? I tested serial by it self and I can communicate to the ESP32:

```
const SerialPort = require('serialport');

const comX = '/dev/ttyUSB0';
const Readline = SerialPort.parsers.Readline;
const port = new SerialPort(comX, {baudRate: 115200});
const parser = new Readline();
port.pipe(parser);
parser.on('data', (dataX)=>{
    console.log(dataX);
});
port.write('\x03\n\r'); //<CTRL-C>

process.stdout.write('hello world');

process.stdin.on('data',(dataX)=>{
    dataX = dataX.toString().trim();
    port.write(`${dataX}\r\n`);
});
```

jakobrosenberg, 2021-09-03:
Closing in favor of https://github.com/pycom/pymakr-vsc/issues/119

closed 2021-09-03 by jakobrosenberg