pycom/pymakr-vsc #150: Pymakr not working in VS code

RoyHopkins, 2021-08-24:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. In visual code installe Pymakr V1.1.12
2. VS code 1.59.1
3.Node JS 16.7.0
4. Windows 64

## What happens?
Pymakr does not show at the bottom of VS code and when I restart VS code I get the message
"There was an error with serialport module, Pymakr will likely noy ..."

## What were you expecting to happen?
To see Pymakr available at the bottom of VS code and to be able to access the Globel commands for Pymakr

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## What versions of software are you using?

**Operating system:**

**VSCode version:**

**Pymakr version:**

maxenium, 2021-08-24:
Same issue here. 

VSCODE
Version: 1.59.1
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-19T11:53:52.479Z (5 days ago)
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Darwin x64 20.6.0

nznobody, 2021-08-30:
Same issue here

```
VSCode:
Version: 1.59.1 (user setup)
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-19T11:56:46.957Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Windows_NT x64 10.0.19042

pymakr: Pymakr v1.1.12
```

jakobrosenberg, 2021-09-03:
Closing in favor of https://github.com/pycom/pymakr-vsc/issues/119

closed 2021-09-03 by jakobrosenberg