pycom/pymakr-vsc #157: New PyMakr version does not open the terminal

William04A, 2021-09-08:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Use a Visual Studio Code version of 1.6.0
2. Upgrade the PyMakr plugin to the newest version, v1.13.0
3. Check the terminal

## What happens?

No PyMakr terminal or commands appear, instead the only available terminals are the ones provided by default by VS Code (Command prompt and PowerShell)

## What were you expecting to happen?

The terminal to open and the PyMakr window to be available.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

From the web debugging tools in VS Code:

```
Activating extension 'pycom.pymakr' failed: Command failed: npx electron-rebuild -v 13.1.8.
$onExtensionActivateError @ mainThreadExtensionService.ts:95
```

## Any other comments?

## What versions of software are you using?

**Operating system:** Ubuntu 20.04 focal

**VSCode version:** 1.6.0

**Pymakr version:** 1.13.0

William04A, 2021-09-08:
If anyone is affected by this issue, I can confirm that downgrading manually to VS Code 1.56 and downgrading the extension (Extensions>PyMakr>Arrow beneath "Disable">Install another version>Install 1.12.0) and then reloading Visual Studio Code fixed it for now.

jalna7, 2021-09-09:
Same problem, reverted to using vscode 1.57 / pymakr 1.1.12 
the auto upgrade to 1.1.13 broke serial port 
I assume but have not tested that 1.59 may work with 1.1.13

jakobrosenberg, 2021-09-09:
**We have release a beta to address this issue. Please see [v.1.1.14-beta.0](https://github.com/pycom/pymakr-vsc/releases/tag/v1.1.14-beta.0).**

For anyone wanting to test this, we would love to get your feedback.

<details>
<summary>How to install a vsix file</summary>

**To install the .vsix file, go to the extensions tab, click the elipsis menu and select `install from VSIX...`**

[image]

</details>

jalna7, 2021-09-10:
Hi, i was supplied the 1.1.14 beta as a vsix and loaded and tested it with vscode 1.60 
it successfully identified the correct serial port loaded the pycom board. i have uploaded some code and seems to be functioning as expected.
may have been specific to my environment ; 
I have 1 or 2 other bluetooth serial ports that present them selves and vscode would often default to these
i always had to problems plugging in a pycom device, i usually needed to restart vscode to find the correct serial port
this release 1.1.14 seems to detect a new serial port being added from the board and rescans and automatically connects.

Tarun75, 2021-09-10:
hi, Beta version not working on my Rpi .Below is the VS code version

Version: 1.58.0
Commit: 2d23c42a936db1c7b3b06f918cde29561cc47cd6
Date: 2021-07-08T06:53:04.182Z
Electron: 12.0.13
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Linux arm 5.10.17-v7l+

CATucker, 2021-09-21:
I've been having this issue for the past month or so and install of 1.1.14-beta.0 resolved this issue for me.

Version: 1.60.1 (user setup)
Commit: 83bd43bc519d15e50c4272c6cf5c1479df196a4d
Date: 2021-09-10T17:07:10.714Z
Electron: 13.1.8
Chrome: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19043

simeneide, 2021-09-23:
Hi, thanks for looking into this. 
I installed 1.1.14beta running on a macbook pro M1 but got the following issue when starting up the extension:

`No prebuilt binaries found (target=14.16.0 runtime=electron arch=arm64 libc= platform=darwin)´

vscode: Version: 1.60.2
OS: Darwin arm64 20.5.0

jakobrosenberg, 2021-09-24:
@simeneide unfortunately we don't support arm yet. If you can cd into your extensions folder, you can try building the bindings manually.

simeneide, 2021-09-25:
ah my bad.
maybe wrong issue, but is there any way of using pycom devices with an m1 mac currently?

jakobrosenberg, 2021-09-25:
Not sure to be honest. @elmasria might know this one.

elmasria, 2021-09-27:
@simeneide maybe you can try Pymakr on Atom

jakobrosenberg, 2021-10-08:
Hi @VirtualWolf I'm unable to reproduce your error. My best advice is to post your issue here: https://github.com/serialport/node-serialport. We use their prebuilt bindings which you can find [here](https://github.com/serialport/node-serialport/releases). You can also try reinstalling VSCode and PyMakr, but I'm doubtful it'll change anything.

satishgaurav, 2022-01-01:
> @simeneide unfortunately we don't support arm yet. If you can cd into your extensions folder, you can try building the bindings manually.

I am getting the following error M1 mac.  
No prebuilt binaries found (target=14.16.0 runtime=electron arch=arm64 libc= platform=darwin). I am currently using Pymakr version 1.1.16. 

Visual Studio Code 
Version: 1.63.2 (Universal)
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Darwin arm64 21.2.0

jakobrosenberg, 2022-01-03:
Hi @satishgaurav. Doesn't the M1 use ARM?

satishgaurav, 2022-01-04:
> 

yes indeed. will wait for the arm version.

jplattel, 2022-01-09:
I'm having the same issue as @satishgaurav, also on a M1, tried different version of both VSCode and Pymakr. Any word on the timeline for the ARM version? Thanks :)

techyporcupine, 2022-02-03:
Hello, I am trying to run this on ARM, and having the same problem as above. I am willing to compile the binaries myself if someone lays out how to go about doing that.

jakobrosenberg, 2022-02-04:
Hi @techyporcupine. Could you try this beta? https://github.com/pycom/pymakr-vsc/releases/tag/v1.1.18-beta.0

  If that doesn't work, Pymakr will try to build the binaries for you. To do that it needs the build tools listed here https://github.com/nodejs/node-gyp#on-macos
> ATTENTION: If your Mac has been upgraded to macOS Catalina (10.15), please read [macOS_Catalina.md](https://github.com/nodejs/node-gyp/blob/master/macOS_Catalina.md).
>
>Python v3.7, v3.8, v3.9, or v3.10
>XCode Command Line Tools which will install clang, clang++, and make.
>Install the XCode Command Line Tools standalone by running xcode-select --install. -- OR --
>Alternatively, if you already have the [full Xcode installed](https://developer.apple.com/xcode/download/), you can install the >Command Line Tools under the menu Xcode -> Open Developer Tool -> More Developer Tools....

techyporcupine, 2022-02-04:
Thanks for your reply, I am still getting 
```
Command 'Pymakr > Global settings' resulted in an error (command 'pymakr.globalSettings' not found)
```
But in VS Code Developer Console, it looks like it is doing more than what it was before: [screenshot from the new beta](https://ibb.co/C1YNmqH)

LeVraiRoiDHyrule, 2022-02-04:
> Thanks for your reply, I am still getting
> 
> ```
> Command 'Pymakr > Global settings' resulted in an error (command 'pymakr.globalSettings' not found)
> ```
> 
> But in VS Code Developer Console, it looks like it is doing more than what it was before: [screenshot from the new beta](https://ibb.co/C1YNmqH)
> 
> Note: I am using Debian 11

Hi, I am having a similar issue. How did you access all these info ? The only info I got is from "runtime status" from VScode extension tab, where the only thing I have is: https://ibb.co/tHBmLdW

I'm on Raspberry Pi 3, Dietpi. It's Raspbian so similar to Debian. Did you find any solution ?

techyporcupine, 2022-02-04:
@LeVraiRoiDHyrule you can go to the Help menu in the top of VS code, and press Developer Tools, once that popup opens, you can press console.

VirtualWolf, 2022-02-05:
No dice on the beta version, I get the same `Command 'Pymakr > Global settings' resulted in an error (command 'pymakr.globalSettings' not found)` error as mentioned above when trying to load Pymakr's global settings, and this in the console:

```
[Extension Host] [PyMakr][serialport] import module...
console.ts:137 [Extension Host] [PyMakr][serialport] import module... failed dlopen(/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/bindings-cpp/prebuilds/darwin-x64+arm64/node.napi.node, 0x0001): tried: '/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/bindings-cpp/prebuilds/darwin-x64+arm64/node.napi.node' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e')), '/usr/lib/node.napi.node' (no such file)
console.ts:137 [Extension Host] [PyMakr][serialport] remove old bindings...
console.ts:137 [Extension Host] [PyMakr][serialport] download bindings...
log.ts:313   ERR [Extension Host] Error: No prebuilt binaries found (target=14.16.0 runtime=electron arch=arm64 libc= platform=darwin)
	at Object.exports.noPrebuilts (/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/prebuild-install/error.js:2:10)
	at /Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/prebuild-install/download.js:77:27
	at FSReqCallback.oncomplete (fs.js:171:23)
```

But doing `cd .vscode/extensions/pycom.pymakr-1.1.18-beta.0 && rm -rf node_modules && npm install` then relaunching VSCode lets me get into the Pymakr settings.

That being said, when I try to actually _connect_ to my ESP32, I get this:

```
mainThreadExtensionService.ts:64 TypeError: SerialPort is not a constructor
	at new PySerial (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/connections/pyserial.js:16)
	at vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/board/pyboard.js:240
	at Function.isSerialPort (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/connections/pyserial.js:113)
	at Pyboard.connect (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/board/pyboard.js:237)
	at vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/pymakr.js:534
	at vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/board/pyboard.js:63
	at vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/main/settings-wrapper.js:125
	at vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/main/settings-wrapper.js:178
	at Function.isSerialPort (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/connections/pyserial.js:113)
	at SettingsWrapper.refreshGlobalConfig (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/main/settings-wrapper.js:176)
	at SettingsWrapper.refresh (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/main/settings-wrapper.js:123)
	at Pyboard.refreshConfig (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/board/pyboard.js:51)
	at Pymakr.continueConnect (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/pymakr.js:476)
	at continueConnect (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/pymakr.js:457)
	at Pymakr.connect (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/pymakr.js:470)
	at PanelView.<anonymous> (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/pymakr.js:74)
	at PanelView.emit (events.js:315)
	at onTermConnect (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/main/panel-view.js:102)
	at Socket.<anonymous> (vscode-file://vscode-app/Users/virtualwolf/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/lib/main/terminal.js:97)
	at Socket.emit (events.js:327)
	at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1132)
```