pycom/pymakr-vsc #161: Please take a look at the startup times

Josverl, 2021-10-05:
# Feature Request 🚀

## Is your feature request related to a problem? Please describe.
Pymakr is slow to start : 
a sample trace : 

```
## Extension Activation Stats

| Extension                             | Eager | Load Code | Call Activate | Finish Activate | Event                             | By                                    |
| ------------------------------------- | ----- | --------- | ------------- | --------------- | --------------------------------- | ------------------------------------- |
| vscode.debug-auto-launch              | true  | 16        | 0             | 21364           | *                                 | vscode.debug-auto-launch              |
| vscode.git                            | true  | 138       | 11            | 21899           | *                                 | vscode.github                         |
| GitHub.codespaces                     | true  | 384       | 18            | 29513           | *                                 | GitHub.codespaces                     |
| mhutchie.git-graph                    | true  | 83        | 4             | 21992           | *                                 | mhutchie.git-graph                    |
| pycom.pymakr                          | true  | 4         | 21295         | 35              | *                                 | pycom.pymakr                          |
```
Note: the 21295 ms really stands out

## Describe the solution you'd like
Startup duration to take significantly less time.

## Describe alternatives you've considered
Currently I (need to) disable pymakr globally to prevent non-MicroPython projects from being slowed down 

**- https://code.visualstudio.com/api/working-with-extensions/bundling-extension**
- https://www.nicoespeon.com/en/2019/11/fix-vscode-extension-performance-issue/
- https://www.freecodecamp.org/news/optimize-vscode-performance-best-extensions/

## Teachability, Documentation, Adoption, Migration Strategy
- quicker startup should improve adoption

elmasria, 2021-10-05:
@jakobrosenberg  please take a look

jakobrosenberg, 2021-10-05:
Hi @Josverl, are you using v1.1.13 or v1.1.14-beta.0?

Josverl, 2021-10-06:
Hi Jacob, 
initial report was on v1.1.13 , 
Just did quick test of the v.1.14-beta, and 

that feels snappier - and the timings also look much better :

```
## Extension Activation Stats

| Extension                               | Eager | Load Code | Call Activate | Finish Activate | Event                                | By                                      |
| --------------------------------------- | ----- | --------- | ------------- | --------------- | ------------------------------------ | --------------------------------------- |
| vscode.debug-auto-launch                | true  | 7         | 0             | 179             | *                                    | vscode.debug-auto-launch                |
| GitHub.codespaces                       | true  | 253       | 22            | 3760            | *                                    | GitHub.codespaces                       |
| madewithcardsio.adaptivecardsstudiobeta | true  | 705       | 2             | 154             | *                                    | madewithcardsio.adaptivecardsstudiobeta |
| mhutchie.git-graph                      | true  | 75        | 5             | 683             | *                                    | mhutchie.git-graph                      |
| ms-python.python                        | true  | 58        | 5             | 799             | workspaceContains:pyproject.toml     | ms-python.python                        |
| ms-python.vscode-pylance                | true  | 207       | 1             | 161             | workspaceContains:pyrightconfig.json | ms-python.vscode-pylance                |
| pycom.pymakr                            | true  | 106       | 128           | 20              | *                                    | pycom.pymakr                            |
| ritwickdey.LiveServer                   | true  | 995       | 8             | 12              | *                                    | ritwickdey.LiveServer                   |

```
That is a **factor**  166 faster 🏎️🏎️ for the 2 samples provided.
If this are the result, just keep removing the code that I monkey-patched in the loader to get things working :-) 

Also notice that the other `eager` extensions finish activation much quicker as well , so there is obviously some interaction there 

I  have my hopes that bundling can push that down even further (below the 20 ms )