pycom/pymakr-vsc #184: installation ov 1.1.16 fails with serialport module under vscode 1.61.2 on ubuntu 21.04

flokain, 2021-11-03:
## What are the steps to reproduce this issue?
1. install pymakr 1.1.16
2.
3.

## What happens?
install fails

## What were you expecting to happen?
installation succeeds w/o errors

## Any logs, error output, etc?
```
[2021-11-03 14:58:35.134] [renderer1] [error] [Extension Host] [PyMakr][serialport] build bindings in Node failed Error: node-gyp failed to rebuild '/home/fkain/.vscode/extensions/pycom.pymakr-1.1.16/node_modules/@serialport/bindings'.
For more information, rerun with the DEBUG environment variable set to "electron-rebuild".

Error: `gyp` failed with exit code: 1
	at NodeGyp.rebuildModule (/home/fkain/.vscode/extensions/pycom.pymakr-1.1.16/node_modules/electron-rebuild/lib/src/module-type/node-gyp.js:109:19)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
	at async ModuleRebuilder.rebuildNodeGypModule (/home/fkain/.vscode/extensions/pycom.pymakr-1.1.16/node_modules/electron-rebuild/lib/src/module-rebuilder.js:94:9)
	at async Rebuilder.rebuildModuleAt (/home/fkain/.vscode/extensions/pycom.pymakr-1.1.16/node_modules/electron-rebuild/lib/src/rebuild.js:226:9)
	at async Rebuilder.rebuild (/home/fkain/.vscode/extensions/pycom.pymakr-1.1.16/node_modules/electron-rebuild/lib/src/rebuild.js:184:17)
	at async prepareSerialPort (/home/fkain/.vscode/extensions/pycom.pymakr-1.1.16/lib/serialport/index.js:49:17)
	at async activate (/home/fkain/.vscode/extensions/pycom.pymakr-1.1.16/pymakr.js:8:5)

```

## Any other comments?
vscode is installed via snap

## What versions of software are you using?

Version: 1.61.2
Commit: 6cba118ac49a1b88332f312a8f67186f7f3c1643
Date: 2021-10-19T14:58:13.605Z
Electron: 13.5.1
Chrome: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.11.0-38-lowlatency snap

pymakr 1.1.16

flokain, 2021-11-03:
i can confirm, that installing same vscode version via.deb package worked. so there must be an issue with snap

jakobrosenberg, 2021-11-11:
Sorry about the slow reply @flokain 

Thanks for the feedback. I didn't know there were issues related to snap, so I'll add that to the readme.

closed 2021-11-11 by jakobrosenberg