pycom/pymakr-vsc #186: Avoid vscode restoring the node.js terminal process on restart

Josverl, 2021-11-11:
# Feature Request 🚀

Vscode tries to revive any terminal proceses that were open on exit.

Apparently this includes the `Pymakr Console` terminal that now gets revived as a `node` terminal 
[image]

this is described here : 
https://code.visualstudio.com/docs/editor/integrated-terminal#_terminal-process-reconnection

## Solution 
- The node terminal not to re-appear after restarting VSCode 
I assume that when the node process is cleanly exited, this behavior is no longer surfaced.

## Alternatives 
It is possible to set the (workplace) settings to never revive terminals / persistent processes .
but it would be nice if this workspace setting was set set directly by Pymakr if it dies not yet exist.

```
"terminal.integrated.persistentSessionReviveProcess": "never"
```

[image]

jakobrosenberg, 2021-11-12:
This has been a pet peeve of mine for quite a while.

My PyMakr console pops up every time I open VSCode, even if I never use it.

Josverl, 2021-11-14:
I think these are 2 different things;
a) vscode by default tries to revive (multiple) terminal sessions that were open, and not closed 
b) pymakr always starts a terminal, even in nodejs workspaces without any python to be seen.

a) can be adjusted by a setting : persistentSessionReviveProcess
b) not as far as I know

closed 2022-04-07 by Josverl