pycom/pymakr-vsc #188: command 'pymakr.globalSettings' not found

dmurray14, 2021-12-09:
## What are the steps to reproduce this issue?
1. Install pymakr extension in VScode using marketplace
2. Restart vscode
3. Command palette: Pymakr > Global settings

## What happens?
Extension does not load. Error message: "Command 'Pymakr > Global settings' resulted in an error (command 'pymakr.globalSettings' not found)"

Same error with any other pymakr related command.

## What were you expecting to happen?
Pymakr extension to load, access to settings

## Any logs, error output, etc?
Error message: "Command 'Pymakr > Global settings' resulted in an error (command 'pymakr.globalSettings' not found)"

## Any other comments?
Have attempted to install vsix package manually, same issue

## What versions of software are you using?

**Operating system:**
macOS 12.0.1 on Apple Silicon (Darwin arm64 21.1.0)

**VSCode version:**
1.63.0 (Universal)

**Pymakr version:**
1.1.16

**VScode info:**
Version: 1.63.0 (Universal)
Commit: 7db1a2b88f7557e0a43fec75b6ba7e50b3e9f77e
Date: 2021-12-07T05:15:48.091Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Darwin arm64 21.1.0

LetsMelon, 2021-12-25:
I have the same problem with an m1 chip

jakobrosenberg, 2022-01-03:
Thanks for the feedback and sorry for the slow reply. Could you try open the global settings again, but this time open the developer tools in VSCode and see if there's a stacktrace logged under `console`.

To open the console:
`Help` -> `Toggle Developer Tools` -> `Console`

dmurray14, 2022-01-10:
@jakobrosenberg yes, a few:

```
mainThreadExtensionService.ts:63 [[object Object]]No prebuilt binaries found (target=14.16.0 runtime=electron arch=arm64 libc= platform=darwin)
$onExtensionRuntimeError @ mainThreadExtensionService.ts:63
```

```
mainThreadExtensionService.ts:64 Error: No prebuilt binaries found (target=14.16.0 runtime=electron arch=arm64 libc= platform=darwin)
	at Object.exports.noPrebuilts (vscode-file://vscode-app/Users/username/.vscode/extensions/pycom.pymakr-1.1.17/node_modules/prebuild-install/error.js:2)
	at vscode-file://vscode-app/Users/username/.vscode/extensions/pycom.pymakr-1.1.17/node_modules/prebuild-install/download.js:77
	at FSReqCallback.oncomplete (fs.js:171)
$onExtensionRuntimeError @ mainThreadExtensionService.ts:64
```
Let me know if you need more detail and I can send the actual logs.

jeffxor, 2022-01-26:
I am experiencing the same issue. Any updates?

cristiantrapero, 2022-01-26:
I am experiencing the same issue in a MacBook Pro 2021 with Apple M1 Pro, MacOS 12.2 Monterey and Vscode version 1.63.2.

[image]

jakobrosenberg, 2022-01-31:
Currently we don't support ARM processors. This is a limitation of https://serialport.io/ which we use to handle USB connections.

The `serialport` library comes with precompiled binaries for various architectures, but not ARM, so the only work around is to install Node dev tools for your system.

I've created a new beta version here that uses NAPI bindings for `serialport` but I can't guarantee that they'll work. https://github.com/pycom/pymakr-vsc/releases/tag/v1.1.18-beta.0

ceedee666, 2022-02-07:
I tried to compile serialport on an Apple M1 Pro to get pymakr running, but didn't succeed. The pymkr documentation states that
> If a working binding can't be downloaded, PyMakr will try to build them locally with node-gyp. Please see requirements for your OS [here](https://github.com/nodejs/node-gyp#installation) 

Therefore I installed node-gyp using `npm install -g node-gyp`. However, the plugin still fails. Is there any documentation what I need to do in order to get the compilation of serialport.io working. As far as I can tell the plugin didn't try to execute node-gyp.

jakobrosenberg, 2022-02-07:
@ceedee and @hallard can I ask you to share the full log from the developers tools. (ctrl + shift + i)

hallard, 2022-02-07:
sure

[image]

jakobrosenberg, 2022-02-07:
That's very peculiar that it ends there. It should attempt to build the bindings after a failed download.

Could you see if you could locate this file on your local machine https://github.com/pycom/pymakr-vsc/blob/develop/lib/serialport/utils/download-prebuild.js#L33 and replace line 33 with `console.log(err)`.

hallard, 2022-02-07:
here it is

If you give me commands to do to manually build I can give it a try, or even share my screen if you want to see, no problem

[image]

jakobrosenberg, 2022-02-07:
I think this is the command to rebuild manually: 

```
npx electron-rebuild -v <electron-version>`
```

For my machine, I believe electron version is `13.5.2`.

hallard, 2022-02-07:
I mean rebuild serialport or pymakr is that correct?
I'll give it a try, run from where ?

jakobrosenberg, 2022-02-07:
It should be at the root of the plugin. It could be something like `~/.vscode/extensions/pymakr`.

hallard, 2022-02-07:
that's what I did, looks like issue with `openssl-fips`

```
charles@m1-office:pycom.pymakr-1.1.18-beta.0$ export  DEBUG="npx electron-rebuild" && npx electron-rebuild -v 13.5.2
⠋ Searching dependency tree  electron-rebuild rebuilding with args: [Arguments] {
  '0': {
    buildPath: '/Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0',
    electronVersion: '13.5.2',
    arch: 'arm64',
    extraModules: [],
    onlyModules: null,
    force: undefined,
    headerURL: undefined,
    types: [ 'prod', 'optional' ],
    mode: undefined,
    debug: undefined,
    prebuildTagPrefix: 'v',
    forceABI: undefined,
    useElectronClang: false,
    disablePreGypCopy: false,
    projectRootPath: '/Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0'
  }
} +0ms
  electron-rebuild rebuilding with args: /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0 13.5.2 arm64 Set(0) {} false https://www.electronjs.org/headers [ 'prod', 'optional' ] false +1ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/bindings +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/binascii +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/consolite +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/copy-paste +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/node-abi +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/prebuild-install +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/rimraf +1ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/serialport +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/bindings/node_modules/@serialport/binding-abstract +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/binding-abstract +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/bindings/node_modules/@serialport/parser-readline +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/parser-readline +1ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/nan +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/debug +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/bindings +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/iconv-lite +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/sync-exec +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/semver +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/detect-libc +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/minimist +1ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/expand-template +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/mkdirp-classic +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/github-from-package +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/napi-build-utils +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/pump +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/rc +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/tar-fs +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/tunnel-agent +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/simple-get +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/npmlog +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/glob +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/binding-mock +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/bindings-cpp +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/parser-byte-length +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/parser-cctalk +0ms
  electron-rebuild exploring /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/@serialport/parser-delimiter +0ms
[... cut ...]
    at async ModuleRebuilder.rebuild (/Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/electron-rebuild/lib/src/module-rebuilder.js:124:14)
    at async Rebuilder.rebuildModuleAt (/Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/electron-rebuild/lib/src/rebuild.js:145:13)
    at async Rebuilder.rebuild (/Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/electron-rebuild/lib/src/rebuild.js:108:17)
    at async /Users/charles/.vscode/extensions/pycom.pymakr-1.1.18-beta.0/node_modules/electron-rebuild/lib/src/cli.js:154:9
```

hallard, 2022-02-11:
Ok I give a new try today, successfully rebuild and after sometime got the editor to open and not firing an error on the plugins. But looks like it can't connect to device, saying also autoconnect is enabled an that's not the case (nor on local project file also) then fire a 'maximum call stack` error

[image]

I'm really stuck with this issue it's an old project where all was working fine and I need only to change little things but unable to get just things working so I'm able to work on.

Any other way to just upload/download files on device without using pymakr?

mortenhauan, 2022-02-19:
Hi, hallard.

Saw this and wanted to give you a tip. Having the same issues as you. Also on an M1 Mac. Pymakr does not work in VSCode. My current workaround is to use Atom with the Pymakr plugin to interact with the board.

Hopefully, a future update will let us do all from VS Code again.

lukich48, 2022-10-03:
It works for me now!
MacOS 12.5 M1
Pymakr v2.22.5
VS Code 1.71.2