pycom/pymakr-vsc #189: Publish to Open VSX

roaldarbol, 2021-12-17:
# Feature Request 🚀
I'd like for Pymakr to publish to [open-vsx.org](https://open-vsx.org/) to be available to alternative VS Code editors such as VSCodium.

## Is your feature request related to a problem? Please describe.
I use VSCodium to avoid tracking and support a project under a FLOSS license. However, as Pymakr is not published on [open-vsx.org](https://open-vsx.org/), it makes installing the extension a lot harder to add.

## Describe the solution you'd like
Publishing to [open-vsx.org](https://open-vsx.org/). No drawbacks, maybe a small amount of work to get started but can even be automated with a [Github action](https://github.com/HaaLeo/publish-vscode-extension#readme).

## Teachability, Documentation, Adoption, Migration Strategy
There's a step-by-step guide [here](https://github.com/eclipse/openvsx/wiki/Publishing-Extensions) about how to publish to Open VSX. And thanks for making this awesome tool by the way - great work!

bandogora, 2022-04-06:
I would appreciate this as an Arch Linux user since the [Community Code - OSS package](https://archlinux.org/packages/community/x86_64/code/) uses Open VSX by default.

roaldarbol, 2022-10-03:
Just pushing this again - it'll not take long, but it will make a difference to a lot of users. :-)

czadikem, 2023-01-28:
Yes, please do this.  This would help everyone greatly.

roaldarbol, 2023-05-16:
While this is still not yet on Open CSX - just a note for other users: You can still just download the `.vsix` file from `Releases` and install from file.

rilstrats, 2023-09-25:
Any updates? This would be awesome!

darltrash, 2024-01-28:
no updates...?