pycom/pymakr-vsc #190: List serial ports

jprodriguez-nbs, 2022-01-03:
# Feature Request 🚀

## List manufacturers when listing serial ports
Any serial port from a manufacturer that is not listed in autoconnect_comport_manufacturers is silently discarded
There are manufacturers (like 'QinHeng Electronics' on some TTGO boards) that do not return an identifiable manufacturer (eg: '1a86'). The user experience is that the pymakr returns always 'Found 0 serialports'

## Describe the solution you'd like
SerialPort.list method used in pyserial.js does return the information about the different serial ports available and has the manufacturer information ready.
The line `var pycom_manu_index = comport_manufacturers.indexOf(manu);` compares the found serial port manufacturer with the configured autoconnect_comport_manufacturers
The solution proposed is to print all found serial ports, both the ones from a manufacturer included in the list and the ones from a different manufacturer, indicating if the manufacturer does match or not.
I would also include a final sentence explaining the user that he can add the missing manufacturers in the pymakr.json autoconnect_comport_manufacturers list if he wants to use that serial port.

## Describe alternatives you've considered
As an alternative, I have manuall listed the ports using nodejs to find the manufacturer, with the following code:

```
const bindings = require('@serialport/bindings')

var listOfPorts=[];

//called automatically by bindings.list()
function list(ports) {
  listOfPorts=ports;
  // now listOfPorts will be the port Objects
  console.log(listOfPorts);
}

bindings.list().then(list, err => {
  process.exit(1)
})
```

If you put this code in test.js, then:
* npm install @serialport/list
* nodejs test.js

You get something like:

```
[
  {
    manufacturer: '1a86',
    serialNumber: '5185015613',
    pnpId: 'usb-1a86_USB_Single_Serial_5185015613-if00',
    locationId: undefined,
    vendorId: '1a86',
    productId: '55d4',
    path: '/dev/ttyACM0'
  },
```

Then you can put '1a86' in the autoconnect_comport_manufacturers list and connect to the board.

jakobrosenberg, 2022-01-04:
@jprodriguez-nbs thanks for the feedback. I'm not sure why USB connections are limited to predefined manufacturers.

Most of all, I'd like to remove the restriction, but I don't know if that could present any side effects.

For now I can add `1a86` to the preapproved devices. You can also submit a PR if you would like.

Josverl, 2022-01-09:
I agree that having a way to add other 
I think that autoconnect could be bad if pymakr starts connection to just any serial device.
to allow some control over it it would make more sense to add a wildcard by default.
that way it would be explicit , and if it causes issues for someone they could just remove that wildcard.
```json 
	"autoconnect_comport_manufacturers": [
		"Pycom*",
		"FTDI",
		"Microsoft",
		"Microchip Technology, Inc.",
		"Silicon Labs",
		"*"
	]
```
I personally would prefer a simple wildcard (above) , but regex (below) would also be fine.
```json 
	"autoconnect_comport_manufacturers": [
		"Pycom.*",
		"FTDI",
		"Microsoft",
		"Microchip Technology, Inc.",
		"Silicon Labs",
		".*"
	]
```

jakobrosenberg, 2022-01-10:
@Josverl I'm currently working on moving the configuration into VSCode settings. This should make it easier to get started without having to remember where everything is.

@jprodriguez-nbs I added your device to the predefined devices, so it should work now without any configuration.

jprodriguez-nbs, 2022-01-30:
Hello,

The board that I am using is a TTGO T-CALL PMU manufactured in 2021 (the serial port to USB device has changed since previous version)

Just for any person that has the same problem, the manufacturer for this USB serial port device in Windows has a different name: "wch.cn", so in order to connect to the device in Windows you have to add "wch.cn" to the autoconnect_comport_manufacturers list

To open the configuration file that has this list use:
* CTRL-P
* Enter ">Pymakr" in the command edit window that appears at the top of the screen
* Select Pymakr global settings
* Modify the list

Example:
```
	"autoconnect_comport_manufacturers": [
		"Pycom",
		"Pycom Ltd.",
		"FTDI",
		"Microsoft",
		"Microchip Technology, Inc.",
		"Silicon Labs",
		"1a86",
		"wch.cn"
	]
```

jakobrosenberg, 2022-04-08:
Closing this for now, but can be reopened if needed.

closed 2022-04-08 by jakobrosenberg