pycom/pymakr-vsc #197: pymakr commands not working - extension not working

joelaustin11, 2022-01-30:
### pymakr.global.Settings not found

## What are the steps to reproduce this issue?
1.  Open VSCode
2. Type Ctrl+Shift+G, to edit global settings
3. error occurs

## What happens?
A VSCode popup says "command 'pymakr.globalSettings' not found" 
The same thing happens with Ctrl+Shift+C to connect to my board.

If i try and execute the command from the command pallete, I get a longer popup, details below

## What were you expecting to happen?
I was expecting to be sent to a new tab and be able to edit the settings of the extension

## Any logs, error output, etc?
longer error output:
`Command 'Pymakr > Global settings' resulted in an error (command 'pymakr.global.Settings not found)`

if i look in `C:\Users\MY_USER\AppData\Roaming\Code\logs\20220130T174522\exthost1\exthost.log` I can see this entry
`[2022-01-30 17:45:24.219] [exthost] [info] ExtensionService#_doActivateExtension pycom.pymakr, startup: true, activationEvent: '*'
`
Which leads me to believe that the extension has started and is running

## Any other comments?
Let me know if there's any info that I didn't give and I will absolutely fetch that

Any help would be greatly appreciated

Thanks!

## What versions of software are you using?

**Operating system:**
Windows 10 - 21H2
intel i5 10th gen 
12GB RAM

**VSCode version:**
VSCode Info:
Version: 1.63.2 (user setup)
Commit: 899d46d82c4c95423fb7e10e68eba52050e30ba3
Date: 2021-12-15T09:40:02.816Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19044

**Pymakr version:**
pymakr v1.1.17

jakobrosenberg, 2022-01-31:
Hi. Can I ask you to share the output from your dev tools console. You can find it here.
[image]

joelaustin11, 2022-01-31:
Hey, today I went to go troubleshoot, and the issue seems to be solved, and i am up and running. 

I think the issue was I had the Pico-Go extension enabled as well for the workspace, so false alarm!

Thanks anyways!

closed 2022-01-31 by joelaustin11

jakobrosenberg, 2022-02-01:
Happy you got it working. 👍

gorgonaut04, 2023-07-21:
I know this is old, but I had the same issue and I think the "fix" was a system restart after installing node. Would recommend adding this step to any troubleshooting docs or FAQ.