pycom/pymakr-vsc #198: Pymakr on Raspberry Pi, Dietpi, arm64

LeVraiRoiDHyrule, 2022-02-04:
## What are the steps to reproduce this issue?
1. Install Dietpi arm64 on Raspberry Pi 3
2. Log in remotely from SSH remote in VScode
3. Install Pymakr in the remote host
4. See the error

## What happens?

Error at Pymakr extension launch : No prebuilt binaries found (target=14.16.0 runtime=electron arch=arm64 libc= platform=linux)

## What were you expecting to happen?

I expect Pymakr to run on the remote Raspberry, so I can control it from VScode on my Windows PC, and executing code on the Lopy4 connected via USB to the Raspberry. (and getting the bottom Pymakr bar to upload from the Raspberry to the Lopy instead of from the PC)

## Any logs, error output, etc?

Error when VScode launches : No prebuilt binaries found (target=14.16.0 runtime=electron arch=arm64 libc= platform=linux)

## Any other comments?

## What versions of software are you using?

**Operating system:** Dietpi

**VSCode version:** 1.64.0

**Pymakr version:** 1.1.17

LeVraiRoiDHyrule, 2022-02-04:
EDIT : Just tried the beta0 for 1.1.18. I get a different error with it:
`The "path" argument must be of type string or an instance of Buffer or URL. Received an instance of Object`

Josverl, 2022-04-06:
[VSCode remote](https://code.visualstudio.com/docs/remote/remote-overview) that you are assuming to work  is currently not supported by either PyMakr v1 or v2