pycom/pymakr-vsc #201: Plugin is not recognizing some MicroPython words

renegadecitizen, 2022-03-20:
the plugin does not recognize some MicroPython words
words like machine or Pin
do not turn colour
I wonder if the plugin is installed right?
can you please help me fix this?

## What versions of software are you using?

**Operating system:**
mac v11.6.5
**VSCode version:**
v1.65.2
**Pymakr version:**
preview of v2

Josverl, 2022-03-20:
For micropython language support I suggest you take a look at https://github.com/Josverl/micropython-stubs

renegadecitizen, 2022-03-21:
Cool thanks 
It looks like it’s command line only
I installed it anyway 
Is there a way of using it in vscode?

Josverl, 2022-03-22:
Working on a simple 'pip install'

jakobrosenberg, 2022-04-08:
@Josverl is this something that could or should be bundled with `pymakr-vsc`? If so, what would the pros / cons be?

Josverl, 2022-04-08:
well mayhaps ,
there are a few approaches, and a few open items.
- as mentioned above I am working to publish the stubs on PyPi. if that is done enabling is as simple as installing the correct stubs from PyPi, and configuring VSCode to run PyLance and optionally pylint.
That is something that PyMakr could help do
- Pico Go did something similar , but includes a specific set of stubs ( for the Pi Pico) with their for of Pymakr v1.
this requires the stubs to be maintained as part of the extension, which brings some drawbacks, but provides a simple installation.

wrt to the stubs itself.
there is quite a variation between :
- MicroPython Families ( Micropython , Pycom , Loboris , and many others )
- Firmware versions ( mpy 1.18 has a differen API that v1.16 , I assume similar for PyCom ) 
- Port - Pycom uses esp32 , but there are many other ports 
- board - different boards have different hardware modules ,
- variants of a board ( ie firmwares optimized / tuned for memory , or OTA or other capabilities) 

in selecting which stubs to install for a project i think the minimum info needed is :
* Family - Version - Port - Board
and based on that do a `pip install pycom-v123_4-esp32-wipy-stubs`  

there are a few ways to create the stubs
- firmware-stubs - generated on the board 
- documentation-stubs - generated from the micropython documentation ( .rst files) 
- core-stubs - re-used from CPython stubs currently published on PiPy

For Pycom: 
- firmware stubs can be geberated with anyone access to the hardware. Im happy to add them to the micropyton-stubs repo and publish them, or you could do this yourself using the micropython-stubber toolset 
- the documentation stubs are a bit harder, as Pycom uses a different documentation format and structure. 
that would require a (perhaps significant) change to the current documentation parser.

Josverl, 2022-04-08:
The same stubs should work for any IDE so expect them to work on Atom as well

jakobrosenberg, 2022-05-24:
Would it be better to move this to discussions?

renegadecitizen, 2022-05-24:
Where to?

Josverl, 2022-06-27:
@renegadecitizen , 

depending on your port/chipset you can get VSCode to recognize Micropython by installing the correct stubs:
 * for esp32 use `pip install micropython-esp32-stubs`
 * for pyboard use `pip install micropython-stm32-stubs`
 * for Pi Pico  use `pip install micropython-rp2-stubs`

ANd indeed lets close or move this to discussions as it is not an issue of pymakr.