pycom/pymakr-vsc #202: Command 'pymakr.__AnyCommand__' not found

redghc, 2022-03-30:
## What are the steps to reproduce this issue?
1. Install vscode
2. Install Pymakr extension from marketplace
3. Execute any Pymkr command (ctrl + shift + p)

## What happens?
Get error "Command 'pymakr.__anycommand__' not found
Modifications are also not shown in the vscode status bar

## What were you expecting to happen?
Execute any command of pymakr

## Any other comments?
NodeJs installed using NVM
Python installed using pyenv

See this issue: https://github.com/pycom/pymakr-vsc/issues/188
However I don't have an ARM processor so I think it's a different problem.

Do the same procedure on an intel i3 processor, getting the same result.

## What versions of software are you using?

**Operating system:**
OS: Arch Linux x86_64 
Host: B450 AORUS ELITE 
Kernel: 5.15.31-1-lts 
DE: Plasma 5.24.3 
WM: KWin 
CPU: AMD Ryzen 3 1300X (4) @ 3.500GHz 
GPU: AMD ATI Radeon RX 470/480/570/570X/580/580X/590 

**VSCode version:**
Versión: 1.65.2
Commit: c722ca6c7eed3d7987c0d5c3df5c45f6b15e77d1
Fecha: 2022-03-10T15:36:26.048Z
Electrón: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.15.31-1-lts

**Pymakr version:**
v1.1.17
v2.8.1 - Preview
(Both fail)

berbert-arthur, 2022-03-31:
Tenho o mesmo problema...

jakobrosenberg, 2022-03-31:
@redghc I just tried installing Pymakr 1 in VSCode and I noticed that it was no longer enabled by default. Would you check that the plugin has been enabled and that your VSCode has been reloaded?

redghc, 2022-03-31:
@jakobrosenberg 

Sure the extension is enabled and restart VSCode.

----
Using the  v2.8.1 - Preview

I noticed the following: 
Opening the debug console get this error.
[image]

Importing "stateManager"

[image]

However the correct name of the file is "StateManager"
[image]

Changing "stateManager" to "StateManager"
In this files

- /src/Device.js
- /src/Project.js
[image]

Now the console error is fixed.

-----

However the main error continues to appear and now there is no error in the console. 

[image]

jakobrosenberg, 2022-03-31:
Good catch on the capitalization. The command manifest was copied from the old Pymakr v1. I think it might be time to clean up all the unused commands now. 😅

I'll push a fix for this and get back to you.

jakobrosenberg, 2022-03-31:
V2.8.2 is out now. I hope this fixes the issues. At least for v2.

berbert-arthur, 2022-03-31:
O processo de terminal "node 'c:\Users\arthu\.vscode\extensions\pycom.pymakr-preview-2.8.2\src\terminal\client.js', 'serial', 'COM2'" foi terminado com o código de saída: 1.
[image]

berbert-arthur, 2022-03-31:
[image]

berbert-arthur, 2022-03-31:
[image]

jakobrosenberg, 2022-03-31:
Thanks @berbert-arthur . This has been fixed in v2.8.3.

berbert-arthur, 2022-03-31:
[Pymakr] Failed to run command: uploadProject. Reason: Traceback (most recent call last): File "<stdin>", line 1, in <module> OSError: [Errno 2] ENOENT. Please see logs for info.

berbert-arthur, 2022-03-31:
[image]

jakobrosenberg, 2022-03-31:
Seems there's an issue with the paths. I'll try get a fix released tomorrow.

redghc, 2022-04-01:
@jakobrosenberg

With the last update some parts of the extension already work (Thank you very much!)

However, there are still some errors. I leave you a summary of those that I have detected. 

configureDevice
-----
[image]

Steps:
[image]
Then select username or password.

[image]
I add some console logs and it seems that it is trying to execute the function menus["username"]  or menus["password"] 

Based on the above and the screenshot below, it attempts to execute a nonexistent function. (Only exist `main` and `autoConnect`)
[image]

eraseDevice
-----
[image]

Select any of the projects and you get these errors 
[image]

I hope this helps in the development of the new version, I would like to support you but I haven't worked on the development of extensions for vscode, I don't have much experience :c .

jakobrosenberg, 2022-04-01:
Thanks much for the feedback!! This is really helpful.

Telnet is currently not supported. The micropython library that we use, does support wireless, but not through a direct telnet connection. I don't know if telnet is a popular feature, but we'll discuss it internally soon. Any feedback on this would greatly appreciated!

The erase feature should be working, so it looks like you found a bug. I'll have a look today.

FWIW this is also my first VSCode extension. Gotta say I love the architecture and API.

jakobrosenberg, 2022-04-01:
@redghc I'm unable to replicate the erase device bug. Do you have any logs you can share?

redghc, 2022-04-01:
The steps to duplicate are:

- Open vscode
- Connect the device with click on the lightning.
- Right click -> Erase device -> Empty project 

Sure here is some screenshoot from developer tools.

[image]

> er, Traceback (most recent call last):
  File "<stdin>", line 20, in <module>
  File "<stdin>", line 6, in rmdir
OSError: -2

> Unable to log remote console arguments Output omitted for an object that cannot be inspected ('TypeError: Cannot read properties of undefined (reading 'getNearest')')

> PyMakr > command > Error: Traceback (most recent call last):
  File "<stdin>", line 20, in <module>
  File "<stdin>", line 6, in rmdir
OSError: -2

-----
I was reviewing the code and I think the problem originates from the execution of this function. 

**Line 83**
[image]

-----
I tried from the "Open terminal log" option, but the file is empty. 
[image]

[image]

I think it has to do with this 

> Unable to log remote console arguments Output omitted for an object that cannot be inspected ('TypeError: Cannot read properties of undefined (reading 'getNearest')')

I hope this helps

jakobrosenberg, 2022-04-04:
Thanks for catching the "open terminal log" misnomer. 👍 It should really be "open terminal history". I'll include a correction in the next patch.

jakobrosenberg, 2022-04-08:
@redghc are you still getting the `erase device` error? I haven't been able to reproduce it locally

redghc, 2022-04-08:
@jakobrosenberg 

If I still have the same error.

[image]

[image]

-----
In fact I found another problem that I think may be related.

If I execute the action "Upload Project"

Get this error:
[image]

[image]

However from what I could notice the project does upload correctly.

Note: This action works fine in the v1.1.8, 

And from what I was able to see, in v1.1.8, the "Upload" action also performs file deletion, which I think is the problem with the "Preview Version" that it can't delete files.

-----

**Note: The action "Download project" works fine.**
But it doesn't throw any notification when it finishes downloading the project, I think it would be good to place some notification that alerts that the download action is finished.

-----

RaspChris, 2022-04-26:
same problem with MacOS 12.3.1, VSC 1.66.2, pymakr v1.1.18
No functional command with pymakr.
Thank you
Chris

jakobrosenberg, 2022-05-02:
@RaspChris are you getting a no functional command error with with Pymakr v2?

RaspChris, 2022-05-02:
Thank you.
We followed this same problem in the Atom section (I tried with both Atom and VSC)  and finally concluded that the board was faulty. I contacted support.

dan-phd, 2022-08-24:
Possibly similar to this issue, I cannot create a project because command "pymakr.createProjectPrompt" not found. Nothing happens clicking the GUI buttons, but I get the error when selecting from the Command Palette.
I am running latest versions of everything (Windows 11 Version 10.0.22000, Visual Studio Code 1.70.2 and PyMakr v2.22.5).

jakobrosenberg, 2022-08-25:
Hi @dan-phd , I expect your install may have been corrupted somehow. Could you try uninstall all `Pymakr` and `Pymakr - Preview` extensions, restart VSCode and then reinstall?

dan-phd, 2022-08-25:
> Hi @dan-phd , I expect your install may have been corrupted somehow. Could you try uninstall all `Pymakr` and `Pymakr - Preview` extensions, restart VSCode and then reinstall?

Thanks @jakobrosenberg however that did not work, I also tried disabling all other extensions and the preview version, still did not work. Then I tried installing a previous version, going back to v2.12.3 worked! More recent versions (2.22.2, 2.22.4 and 2.22.5) do not. Hopefully there is scope now to see what changed between 2.12.3 and 2.22.2 ?

markusfloer, 2022-09-01:
[image]

markusfloer, 2022-09-01:
> [image]

I have the latest release of pymakr installed on vscode. Trying to upload a new project to gpy device, Iæm not allowed to press the upload button, so trying to delete existing project. This is the error message that appears.

jakobrosenberg, 2022-09-05:
Hi @dan-phd . Apologies. I completely missed your last reply. If you switch to the preview version, you should be able to pinpoint the exact patch that broke your setup. From there I can have a look and see what happened.

@markusfloer that also looks like the extension didn't install properly. If previous versions work for you, I should be able to fix it - granted that you're able to identify the last working version. I suspect it's the same issue you're facing.

mp3user, 2022-12-01:
Fresh install latest extension is work but after I disable and then enable it and now it completly broken.
Any command is failed with "command pymekr.name not found".
Ive try reinstall extension with older version and vscode as well

aka13-404, 2023-03-26:
Happe

> Possibly similar to this issue, I cannot create a project because command "pymakr.createProjectPrompt" not found. Nothing happens clicking the GUI buttons, but I get the error when selecting from [...]

Happens to me as well. Preview seems to work though.