pycom/pymakr-vsc #204: There was an error with your serialport module

schmt16, 2022-03-31:
## What are the steps to reproduce this issue?
1. Win10 x64
2. VScode 1.65 with Pymakr 1.1.17
3. VScode update to 1.66

## What happens?
Pymakr error on serial port bindings

## What were you expecting to happen?
Pymakr working like before update

## Any logs, error output, etc?
"There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)"

## Any other comments?
Downgrade to VScode 1.65 solved the issue temporarily.

## What versions of software are you using?

**Operating system:**
Win10 x64

**VSCode version:**
VScode 1.65, 1.66

**Pymakr version:**
Pymakr 1.1.17

redghc, 2022-03-31:
I had the same problem a few days ago.
Make sure you have Node.js installed, this fixed it for me

https://nodejs.org/en/download/

schmt16, 2022-04-01:
@redghc Yes, i have instelled node.js. I think, this is a basic requirement for the pymakr vscode extension. As a vscode dev explained to me, the vscode update changes the node.js configuration, which has to be implemented by the extension developers...
https://github.com/microsoft/vscode/issues/146418#issuecomment-1084742575

jakobrosenberg, 2022-04-01:
@schmt16 can you try
- uninstall Pymakr
- reload VSCode
- install Pymakr

There's also a preview of Pymakr 2. It should work far better with VSCode, but still has a few bugs and doesn't feature telnet.
https://marketplace.visualstudio.com/items?itemName=pycom.pymakr-preview#review-details

schmt16, 2022-04-01:
@jakobrosenberg Thanks. Pymakr works perfect after downgrading to VScode 1.65 yesterday

Now:
- uninstall pymakr
- update VScode to 1.66, restart
- install pymakr
- get the same serialport errors as yesterday (vscode console output attached)
- reinstall vscode 1.65
- pymakr works without errors

[errors.txt](https://github.com/pycom/pymakr-vsc/files/8395806/errors.txt)

[image]

Potatotatotato, 2022-04-03:
@schmt16  You are right. The VS Code 1.65 does not support Pymakr, which boders me 5 hours.

Potatotatotato, 2022-04-03:
@schmt16 You are right. The VS Code 1.65 does not support Pymakr, which costs me 5 hours to configure the environment.

Josverl, 2022-04-07:
Pymakr v1.18 has been released yesterday which solves this issue.
Also the SerialPort module (the smoking gun in most of these breaks) has been updated to support a new Electron integration model (N-API) that _should_ prevent this the same break  in a future VSCode update.

jakobrosenberg, 2022-05-24:
Closing this as issue should be resolved.

closed 2022-05-24 by jakobrosenberg