pycom/pymakr-vsc #205: Open device in File Explorer fails (on windows)

Josverl, 2022-04-02:
## What are the steps to reproduce this issue?
1. install pymakr v2.x
2. connect to a board :⚡
3.Open device in File Explorer 

## What happens?

4. File Explorer shows a device : `serial:/COM4 `, with an **`!`**  
5. hover over the ! shows `\flash * unable to retrieve workspace folder`

## What were you expecting to happen?
list of files on the device 

## Any logs, error output, etc?

developer log shows 
``` log 
[Extension Host] listening on 5364
console.ts:137 [Extension Host] PyMakr > updated config: Object
console.ts:137 [Extension Host] PyMakr > Device: Silicon Labs CP210x USB to UART Bridge (COM4) > connecting...
console.ts:137 [Extension Host] PyMakr > Device: Silicon Labs CP210x USB to UART Bridge (COM4) > connected.
log.ts:313   ERR No file system provider found for resource 'serial://com4/flash': ENOPRO: No file system provider found for resource 'serial://com4/flash'
    at p.withProvider (vscode-file://vscode-app/c:/Users/josverl/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:610:27776)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async p.exists (vscode-file://vscode-app/c:/Users/josverl/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:610:30371)
    at async Promise.all (index 1)
    at async vscode-file://vscode-app/c:/Users/josverl/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:2110:2110
mainThreadExtensionService.ts:107 Activating extension 'vscode.npm' failed: No file system provider found for resource 'serial://com4/flash'.
$onExtensionActivationError @ mainThreadExtensionService.ts:107
console.ts:137 [Extension Host] rejected promise not handled within 1 second: Error: ENOENT: no such file or directory, scandir 'C:\flash'
D @ console.ts:137
console.ts:137 [Extension Host] stack trace: Error: ENOENT: no such file or directory, scandir 'C:\flash'
D @ console.ts:137
log.ts:313   ERR ENOENT: no such file or directory, scandir 'C:\flash': Error: ENOENT: no such file or directory, scandir 'C:\flash'
log.ts:313   ERR [Extension Host] (node:8856) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.

```

## Any other comments?

## What versions of software are you using?

Window:  11

VSCode: 
Version: 1.66.0 (user setup)
Commit: e18005f0f1b33c29e81d732535d8c0e47cafb0b5
Date: 2022-03-30T05:50:14.623Z
Electron: 17.2.0
Chromium: 98.0.4758.109
Node.js: 16.13.0
V8: 9.8.177.11-electron.0
OS: Windows_NT x64 10.0.22572

**Pymakr version: 2.8.4**

jakobrosenberg, 2022-04-06:
I haven't been able to solve this one yet. I get the same behavior, but for some reason it still works.

Josverl, 2022-04-13:
I think this is related to the MCU root
As discussed in 
https://github.com/pycom/pymakr-vsc/discussions/224

jakobrosenberg, 2022-05-16:
From what I can tell, whenever a folder is mounted on a new protocol, like `serial://`, VSCode restarts. It happens subtly and is easy to miss.

I had a look at VSCode's official memfs example last week and it seems this always happens on the first mount.

I'll see if I can allocate some more time for it this week.

@Arcadia197 I suspect the `/flash` path is a different, but related issue. I've added it to the todo list.

Josverl, 2022-05-16:
> I had a look at VSCode's official memfs example last week and it seems this always happens on the first mount.
Can you share a link ?
if this is in the sample then it may be a omission / regression and should be reported upstream

jakobrosenberg, 2022-05-16:
Hi @Josverl, you can see the samples [here](https://github.com/microsoft/vscode-extension-samples). It's the [fsprovider-sample](https://github.com/microsoft/vscode-extension-samples/tree/main/fsprovider-sample).

Granted my observations are correct, I think the restart is by design.

Josverl, 2022-05-16:
thanks , ill have a look later this week to see if I can reproduce

jakobrosenberg, 2022-05-19:
I think I've narrowed down the issue. It's not related to first time use of a protocol after all.

VSCode restarts when the primary workspace (1st) is changed and/or when the workspace change from single to multi.

I'll see if I can find a solution / workaround.

jakobrosenberg, 2022-05-19:
Just tried this in the `memfs` example:
```javascript
context.subscriptions.push(vscode.commands.registerCommand('memfs.workspaceInit', _ => {
    vscode.workspace.updateWorkspaceFolders(1, 0, { uri: vscode.Uri.parse('memfs:/foo'), name: "MemFS - Sample" });
}));
```

Got the following
```
rejected promise not handled within 1 second: Error: ENOENT: no such file or directory, scandir 'C:\foo'
c:\Users\jakob\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:102
stack trace: Error: ENOENT: no such file or directory, scandir 'C:\foo'
```

Seems that any unresolved path is interpreted as a local filepath.

Josverl, 2022-05-22:
I indeed see a correlation between the existence of a local folder `C:\flash` and better behavior
But that folder should not need to be created, so perhaps some of the functionality does not use the virtual file system but the actual.
I also found that `log.warn` does not log warnings, so we may be missing some telltales.
Raised a separate issue for that

jakobrosenberg, 2022-05-23:
Sorry. I completely forgot to follow up on this. The issue is caused by VSCode restarting whenever the primary workspace changes or when the workspace changes from single to multi.

I also tried this path in the memfs example: `memfs:/foo` and got an error about `c:\foo` missing. It seems this issue is caused by VSCode wrongly normalizing the error on Windows machines.

Josverl, 2022-05-24:
I think this is now mostly solved by (manually) changing the root path configuration to match the device and other changes 
one think that remains confusing for first time users is that pymakr signals that the MCU is not responding and offers to restart it in safe mode (only possible on some hardware / firmwares) 

In my testing this seems occurs every time when mounting a device creates a new workspace, and then the tendecy is to click that [safe-boot] button , and in my case that always makes the problem bigger as I do not have a device that supports safe-boot 

I'll do some more testing , and if needed create a issue to tackle that behaviour

closed 2022-05-24 by Josverl

jakobrosenberg, 2022-05-24:
I have considered changing the root path, but I'm not sure if there are any downsides. For instance what happens if the user has both `/flash` and `/sd`. Then again, that might actually be a case for leaving `/` as the root path instead.

DOBsby, 2024-09-08:
I've just run into this issue. How do I change the root path configuration? I'm looking at the line 

"terminal.integrated.cwd": "/"

in

settings.json, is that it? And how would I make it "match the device and other changes"? Sorry if this is a bit basic.

Josverl, 2024-09-09:
The company pycom does not exist any more, and therefore there is no maintenance to pymakr anymore. 
You could see if [MicroPico](https://marketplace.visualstudio.com/items?itemName=paulober.pico-w-go) may be a relevant option.
It is based off Pymakr , and allows for Pico and other devices.