pycom/pymakr-vsc #212: vNext: terminal forces ESPxxx boards into bootloader on ardware reset

Josverl, 2022-04-06:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. install vNext
2. connect ESP32 board to serial 
3. Connect to the serialport and open a Pymakr Terminal
4. press hardware reset on the boar 

## What happens?
- the board reboots and enters the bootloader to download new firmware
- 
## What were you expecting to happen?
- hardware reboot and reconnection to the exiting terminal ( behaviour in v1 ) 

## Any logs, error output, etc?

```
MicroPython v1.17 on 2021-09-02; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
[[[ Press hardware reset ]]]
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x7 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

```

## Any other comments?
I fixed this same issue in v1 by setting dtr and rts high according to the ESP docs 

```
    if (process.platform == 'win32') {
      // avoid MCU waiting in bootloader on hardware restart by setting both dtr and rts high
      this.stream.set({ rts: true });
    }
```

## What versions of software are you using?

**Operating system:** Windows 11 
**Pymakr version:** Vnext

Josverl, 2022-04-07:
@jakobrosenberg , note that it is worth the effort to test if the same fix would be needed on mac/linux, or if that the serialport interface  behaves differently these by convention. I just fixed it for windows as there were no reports of the same on other platforms, and I had no option to test ....

jakobrosenberg, 2022-04-25:
Hi @Josverl . Did we merge your fix?

Josverl, 2022-04-25:
That fix has been in V1 since some 4 years: 
- https://github.com/pycom/pymakr-vsc/issues/16
- https://github.com/pycom/pymakr-vsc/pull/19

For V2 the serialport interface has been `outsourced` to micropython-ctrl-cont for which I raised this PR as I think you have seen.
 - https://github.com/jakobrosenberg/micropython-ctl/pull/1

Josverl, 2022-05-19:
Thanks for merging https://github.com/jakobrosenberg/micropython-ctl/pull/1,
I did a quick test using the updated module and generic ESP32 and ESP8266 boards and that indeed avoids the MCU to wait in the bootloader

`.vscode\extensions\pycom.pymakr-preview-2.13.0> npm install micropython-ctl-cont `
then connect and reset 

``` python
MicroPython v1.18 on 2022-01-17; ESP32 module with ESP32
Type "help()" for more information.
>>> import machine;machine.reset()
WebREPL is not configured, run 'import webrepl_setup'
MicroPython v1.18 on 2022-01-17; ESP32 module with ESP32
Type "help()" for more information.
>>> 
```
also verified to work for the `hardware reset` menu option.

when `micropython-ctl-cont@1.13.10` has been merged into V2 this can be closed.

closed 2022-05-24 by Josverl