pycom/pymakr-vsc #213: use the vscode output channel for pymakr logs - vNext

Josverl, 2022-04-06:
# Feature Request 🚀

## Is your feature request related to a problem? Please describe.
to report/troubleshoot probblem end-users are directed to the developer debug logs in vbscode.
while this works, somewhat, these are not intended to be used by end-users.

better yet, there is functionalitly in vscode that can be accessed with just a few lines of code

## Describe the solution you'd like
```
const createLogger = (name) => {
  //Create output channel
  let outputChannel = vscode.window.createOutputChannel("PyMakr");
  //Write to output.
  outputChannel.appendLine("this is your user friendly logger.");
  
  const log = _createLogger(
    {
      methods: {
        debugShort: console.log,
        traceShort: console.log,
        info: outputChannel.appendLine, // :-( does not work this simple

      },
    },
    name
  );
```

However I could not find how to persuade `consolite` to call [`outputChannel.appendLine`](https://code.visualstudio.com/api/extension-capabilities/common-capabilities#output-channel) rather than `console.log`

## Teachability, Documentation, Adoption, Migration Strategy

[image]

jakobrosenberg, 2022-04-07:
Thanks!! I've been looking for that.

You would need to bind the method for it to work
```javascript
info: outputChannel.appendLine.bind(outputChannel)
```
or you could do this
```javascript
info: (...params): outputChannel.appendLine(...params)
```

The reason it works for console.log is that the custom methods get mounted in an object that's already bound to console.

Josverl, 2022-04-07:
binding seems to work , 
but still it does not appear to be called /integrated by consolite, and im not sure how to trace/debug that 
(see above PR) 

I think the 2nd example may be typescript , at least that is what code thinks :-)

jakobrosenberg, 2022-04-08:
My bad. I had accidentally written a `:` instead of a `=>`. 😅

jakobrosenberg, 2022-04-11:
I found the issue. `appendLine` only accepts a single string, while `console[method]` supports a variadic input of any type. The solutions was

https://github.com/pycom/pymakr-vsc/commit/4b4b8a1015234fe6509674193decdd7169d7d788#diff-2ec4d3a277ed3e946eafeaa0ad4a72485817306f3afbf294a3d0d34adfd3e947R16-R19

I have merged your PR with the changes into next-staging.