pycom/pymakr-vsc #230: Cannot reopen pymakr termianl when accidentally closed

scath999, 2022-05-06:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Install Pymakr for VS-Code 1.1.18
2. **Notice:** A Pymakr Terminal is available in the Terminal Window
3. Close the Pymakr Terminal window by clicking on the trashcan icon beside it's name in the list of open terminal consoles
4. Click the '+' sign at the top of the Terminal Window

## What happens?
**Notice:** There is no option to open a new Pymakr Terminal.

[image]

## What were you expecting to happen?
Expected Outcome: Pymakr Terminal is included in the list of available terminals in this menu

## Any logs, error output, etc?
None.

## Any other comments?
Please feel free to ask for further information, or direct me towards a solution - I can't find one.

## What versions of software are you using?

**Operating system:**
Win 10

**VSCode version:**
1.66.2
**Pymakr version:**
1.1.18
** Node Version:**
16.13.0

jakobrosenberg, 2022-05-10:
As far as I know, this is by design for v1. I recommend upgrading to v2. There's a preview of it in the VSCode marketplace.

I'm closing this issue for now as this is addressed in v2. If you have questions, feel free to reply.

closed 2022-05-10 by jakobrosenberg

scath999, 2022-05-10:
Understood,  and thanks for a point in the right direction. I'll take a look, but happy to assume the issue is closed for now.

Many thanks 😊