pycom/pymakr-vsc #232: Clicking sync/upload does nothing and device gets stuck in busy state only when using pymakr.

duemoselund, 2022-05-15:
The device is recognized and a terminal output is printed out clicking download does correctly download the files from the device. However, clinking sync or Upload To Device to upload any new changes does nothing.
Instead, I have to use the pkymakr -> Run On Device -> Click Ok to the already selected device in the drop-down list.

Furthermore, after aprx. 5 minutes the terminal stops printing output and pmkakr reports the device is busy and can't reach it. Trying a soft/hard does nothing and gives the same message of being busy, and the device has to be physically reconnected. 

Switching to Thonny to test if it was a hardware issue shows no issues. Everything uploads fine, and the output runs indefinitely.

VSCode: Version: 1.67.1
OS: Linux x64 5.17.6-200.fc35.x86_64
Pymakr: 2.12.13
ESPWROOM32

[image]

[image]

[image]

jakobrosenberg, 2022-05-16:
Hi @duemoselund , thanks for reporting this. 

> clinking sync or Upload To Device to upload any new changes does nothing.

1. Does the device show as busy (the lock) before you click upload?
2. By "does nothing", do you mean that the new files don't run after you upload them or that the files aren't uploaded at all?

> Furthermore, after aprx. 5 minutes the terminal stops printing output and pmkakr reports the device is busy and can't reach it. Trying a soft/hard does nothing and gives the same message of being busy, and the device has to be physically reconnected.

Could you provide the exact errors that you receive? It makes it easier for us to discern whether the errors were thrown by Pymakr or one of its libraries.

hidara2000, 2022-05-17:
https://user-images.githubusercontent.com/15170494/168710236-69b1b0cc-15f3-45b5-b19e-32b14208f583.mp4

Similar issue here. Can't upload any new code. Downloading works fine. See attached video. Thonny works perfectly

jakobrosenberg, 2022-05-17:
Thanks for the video @hidara2000 . What board and firmware are you using?

Random-Cow, 2022-05-17:
I too have the same problem. I'm also using an ESPWROOM32 but on Windows instead of Linux.
However i was able to fix it by changing the default values for the rootPath in the settings.json file (see [here](https://github.com/pycom/pymakr-vsc/discussions/224) when using the latest preview)

instead of 
`        {
            "match": "manufacturer=.*",
            "field": "rootPath",
            "value": "/flash"
        }`
I changed it to:
`        {
            "match": "manufacturer=.*",
            "field": "rootPath",
            "value": "/"
        }`
I could have probably also created a new entry to match the manufacturer but for me it works 🙂.
On a side-note that setting also fixed the "open device in file explorer" for me, so probably it is also related.

Josverl, 2022-05-17:
@Random-Cow , Indeed both are related to the differences of the rootPath used by the different MicroPython Families 
I find the below (user) config works decently for both Pycom and Standard firmwares ( assuming that Pycom Hardware uses Pycom Firmware) 

vscode: `ctrl-,` `pymakr.devices.config` 
```json
    "pymakr.devices.config": [
        {
            "match": "manufacturer=.*",
            "field": "rootPath",
            "value": "/"
        },
        {
            "match": "manufacturer=Pycom.*",
            "field": "rootPath",
            "value": "/flash"
        }
    ]
```

jakobrosenberg, 2022-05-18:
I'm closing this for now, but feel free to keep the conversation going. If the OP's issue is unrelated to the `rootPath` solution, I can reopen the issue.

closed 2022-05-18 by jakobrosenberg