pycom/pymakr-vsc #238: V2: The different **device lists** should use the same names and filters

Josverl, 2022-05-21:
using multiple devices is great , but can be confusing when using multiple devices , and across platforms as the device naming currently is not consistent.

   - Devices Pane :
      - custom name or default name 
      - filters used 
   - Project > Selct Devices 
     - name : raw port name 
     - filter : not used 
   - folder > menu > Pymaker > Upload to device [Ctrl-Shift-S]
     - name : raw port name 
     - filter : not used 
   - run file/selected code on device [[Ctrl-Shift-R]
      - name : raw port name 
       - filter : not used 
       - no default ( even when the file is on the device 

I would prefer consistent device naming:

For serial ports using multiple devices is great, but can be confusing when using multiple devices , and across platforms as the device naming, ordering , filtering and default selection currently is not consistent.
also, if only one device is connected, then it makes sense to skip the selection prompt altogether.

   - Devices Pane :
      - custom name or default name 
      - filters used 
   - Project > Selct Devices 
     - name : raw port name 
     - filter : not used 
   - folder > menu > Pymaker > Upload to device [Ctrl-Shift-S]
     - name : raw port name 
     - filter : not used 
   - run file/selected code on device [[Ctrl-Shift-R]
      - name : raw port name 
       - filter : not used 
       - no default ( even when the file is on the device 

I would prefer consistent device naming:

for serial ports it makes sense to start with the port name on all platforms (is currently different on windows / linux ) 
I would suggest something like
 * `serial://port/<name>[/project]`
 * `telnet://address:port/<name>[/project]`
 * `webrepl://address:port/<name>[/project]`

jakobrosenberg, 2022-05-24:
I much agree. Naming should definitely be consistent and always fall back to the raw port name if no custom name is provided. Come to think of it, this should all be handled through the config.

[image]

The `protocol` variable was missing from the template, but has been added to `next-staging`. Now I just need to make names consistent.

jakobrosenberg, 2022-05-24:
Fixed in https://github.com/pycom/pymakr-vsc/commit/bade6d5daa1ae62502c5fdc9ae06f39c63b0f1f6 and https://github.com/pycom/pymakr-vsc/commit/e3d6c76c3e410a7b6b205a8c42be466a6cce5a80

closed 2022-05-24 by jakobrosenberg