pycom/pymakr-vsc #242: Can't upload project

pgrenon1, 2022-05-24:
Board: NodeMCU with ESP-8266 (CH340g)
TLDR: I'm trying to upload scripts from Visual Studio Code to this board using Pymakr and it's not working.

I followed the steps at https://docs.micropython.org/en/latest/esp8266/tutorial/intro.html in order to flash micropython on the board.
Here is what I did, steps by steps:
* connected the device and loaded micropython on it successfully
    * I used this version https://micropython.org/download/esp8266/ since the board has 4mb flash, but to be sure, I also tested with the 1mb version and with the 512kb version. same result.
* opened VS code with Pymakr plugin installed and created a pymakr project
    * Note: at this point, I can write python in the console, but when I click the "open device in file explorer" I get this: 
[image]
* modified the code inside ```boot.py``` to add a connect method, following the tutorial here https://docs.micropython.org/en/latest/esp8266/tutorial/network_basics.html
* I save the changes and hit this button to upload: 
[image]
* but when I click the download button here, my code has disappeared from ```boot.py``` 
[image]

Also, if I add a file ```main.py``` to the project and press the "Sync project to device", nothing is uploaded because I only see this!
[image]

Can anyone help me with this please?

Thanks in advance

nkgiovannivl, 2022-05-27:
Having same issue for me..

jakobrosenberg, 2022-05-30:
This was answered here: https://forum.pycom.io/topic/7576/esp8266-can-t-upload-project/2?_=1653891246201

**Solution:**
1. Open settings and type pymakr
2. Goto Pymakr > Devices: Config
3. Update `/custom/path` to match the root folder of your device. Make sure to also update the `match` field. You can match against any property here. (for a full list of properties, hit <kbd>F1</kbd> and run the `list devices` command)

The next patch *may* auto detect root path so the above steps won't be required.

I'm closing this, but feel free to continue the conversation. 🙂

closed 2022-05-30 by jakobrosenberg

LittleHobbyShop, 2022-05-31:
@jakobrosenberg Would you please share some screenshots of what you've described with editing the settings?

I'm having the same issue as described but with ESP32 and suspect it's a root folder problem but I can't find Devices: Config in settings.

LittleHobbyShop, 2022-06-01:
OK, thanks for that @jakobrosenberg that's got me started however I now have VS Code not recognising that as a setting??

[image]

My settings text is this, I can't see anything wrong with it compared to yours above

` "pymakr.devices.config": [
        {
            "match": "manufacturer=Silicon Labs",
            "field": "rootPath",
            "value": "/"
        },
    ]
`

The autocomplete only gives me these options:
[image]

pgrenon1, 2022-06-02:
Also @jakobrosenberg I was told on the Pycom forum that my board (ESP8266) is not supported for pymakr and that was why I couldn't upload. Is that a mistake?

AussieSusan, 2022-06-09:
Looking at the screen shots, I can get to the point of selecting the 'Devices' item but there is no 'Edit is settings.json' option.
The options I have are "Auto Connect", "Include", "Names" and "Name Template". 
Is there some other version I should be installing???
Susan

jakobrosenberg, 2022-06-10:
@AussieSusan If you upgrade to the [latest preview](https://marketplace.visualstudio.com/items?itemName=pycom.pymakr-preview), 3rd party devices should now be auto detected.

@pgrenon1 yes and no. Pymakr is for Pycom devices, but we do try to make it work with 3rd party devices. There's a fix in the latest preview version that auto detects the correct root path on 3rd party devices. I don't have any 3rd party devices to test, so would love to hear your experience if you try it.

pgrenon1, 2022-06-10:
> @AussieSusan If you upgrade to the [latest preview](https://marketplace.visualstudio.com/items?itemName=pycom.pymakr-preview), 3rd party devices should now be auto detected.
>
> @pgrenon1 yes and no. [...]

Hey @jakobrosenberg I upgraded to the preview using the link you sent earlier and it did not detect the root path. But I haven't tried doing the whole device setup I just created a new blank pymakr project.

I was however able to go in the settings and change it so it matches my device and now everything seems to work!

grzegorzwozny, 2023-11-03:
I had a similar problem with code uploading in Pymakr v2.22.5.

I solved this problem by using the CP2102 USB<->UART adapter instead of the built-in CH340.

[image]