pycom/pymakr-vsc #245: MacOS iMac M1 - Not installing or working correctly

AussieSusan, 2022-05-31:
I've installed the ARM version of VSCode V1.67.2 on my iMac M1 (see below). However I can't seem to get many extensions to work. For example, I install 'PyMakr', the only command available is "PyMakr: Focus on devices view. There are no options in the bottom bar. When I try to run a program it runs locally and I can't see any way to upload the multiple files that make up my project.
However in the 'Devices' list, I can connect a terminal to the board so some parts seem to be working.
(On the other hand Espressif and PlatformIO seem to work!) This seems to be a fundamental issue with my installation rather than an extension-specific issue so any assistance would be appreciated. 
Susan

Version: 1.67.2 Commit: c3511e6c69bb39013c4a4b7b9566ec1ca73fc4d5 Date: 2022-05-17T18:20:04.972Z Electron: 17.4.1 Chromium: 98.0.4758.141 Node.js: 16.13.0 V8: 9.8.177.13-electron.0 OS: Darwin arm64 21.5.0

jakobrosenberg, 2022-05-31:
Hi @AussieSusan . Can you tell me what version of Pymakr you're using?

AussieSusan, 2022-05-31:
V2.12.3
I believe this is the latest in the Extension Marketplace - it was the default version when I tried to install it (again) yesterday just before posting.
Susan

jakobrosenberg, 2022-06-01:
For v2, Pymakr has been moved to it's own view instead of the bottom bar.

If you haven't already, please have a look at [getting started](https://github.com/pycom/pymakr-vsc/blob/next/GET_STARTED.md).

AussieSusan, 2022-06-04:
Thanks for the pointer - that helped but I still can't get the code to upload to the device.
The first issue was that I created the project but when I went to the PyMakr window to select the device, it showed other projects but not the new one. I eventually got that to work (by using the '+' sign and selecting the existing project) and then selected the USB serial device and clicked on the 'Sync project to device' icon. Nothing happened.
I then tried the 'Hard Reset' option and that did reset the board. I then tried the 'Sync..." icon again and it went into 'Raw REPL; CTRL-B to exit' and was stuck on the ">" prompt. Ctrl-B did bring back the standard REPL prompt but the codw was not uploaded.
What is happening now is that the board will sit at the REPL prompt for a sort time and then, by itself, drop into the 'Raw REPL' mode. This occurs even after rebooting the board (by pressing the 'Boot' button). Running another 'Hard Reset' seems to stop the cycle.
Susan

jakobrosenberg, 2022-06-08:
If you're using a third party device, please have a look at this https://github.com/pycom/pymakr-vsc/issues/242#issuecomment-1140751462.

AussieSusan, 2022-06-09:
Continued over in the other site you linked to.
Susan

jakobrosenberg, 2022-06-10:
Thanks Susan. I'll close this issue.

closed 2022-06-10 by jakobrosenberg