pycom/pymakr-vsc #248: [BUG] PyMakr 2 menus visible in/on other applications

Josverl, 2022-06-02:
**Describe the bug**
The device `...`  menu item and submenu are visible in several other locations 
I think this is: `pymakr.projectDeviceMenu` 

I think this is caused by a too broad visibility of that menu:
``` json 
        {
          "submenu": "pymakr.projectDeviceMenu",
          "when": "viewItem != project",
          "group": "inline@100"
        },
```

I think the visibility may be better expressed as `when": "viewItem == device || viewItem == projectDevice",`

**To Reproduce**
Steps to reproduce the behavior:
1. install Pymakr 2
2. install other extensions such as `Python environment manager`, `remote - SSH`
3. view a treeview / explorer in another extension 
4. Click on an item '....'
6. See the pymakr device menu showing up

**Expected behavior**
Pymakr not to pollute other extensions 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win 11 X64
 - Pymakr : v2.14.0
 - VSCode: 1.67.2
 - Device : n/a

jakobrosenberg, 2022-06-02:
Great catch @Josverl . I've never come across any leakage in any of my plugins, but then again I always run it in debug mode with other extensions disabled.

VSCode has an `extension` field to match against in the `when` field, but for some reason I can't get it working.

Prefixing the when conditions with `extension =~ /.*/ && ` works, but `extension =~ /.*pymakr.*/ && ` doesn't.

Josverl, 2022-06-03:
I think I see that you already solved the when by restricting it to a pymakr view.

on the regex: perhaps there is no `extension` property on the treeItem  to match against.

jakobrosenberg, 2022-06-08:
Did some regex testing and found out that there's a property but the value is set to "undefined". It seems odd that we can't restrict our buttons etc to the plugin itself. At best we can limit by view and viewItem names.

Using view names would mean that we have to do
```
"view == pymakr-devices-tree || view == pymakr-projects-tree && ..."
```
or
```
"view =~ /^pymakr-/ && ..."
```
I like neither. The first is messy and the second imposes a performance hit. Furthermore using the view property seems like a bad practice as it can still conflict with other extensions - even for us, if we have the both the stable and preview version installed.

Alas, I think this will be the solution `"view =~ /^pymakr-/ && ..."` unless there's some way to get the `extension` value working.

jakobrosenberg, 2022-06-08:
Opened an issue here. https://github.com/microsoft/vscode/issues/151493

jakobrosenberg, 2022-06-16:
This should be fixed now. https://github.com/pycom/pymakr-vsc/commit/ccc9efb843ccc5ab99b1e1c739a4c7b53016cbd8

closed 2022-06-16 by jakobrosenberg