pycom/pymakr-vsc #253: [BUG] Nothing at all happens when clicking "Sync project" button on M1 Mac

VirtualWolf, 2022-06-05:
# Describe the bug
I'm able to get Pymakr to connect to my board and bring up a REPL window (though every time I disconnect or reconnect I have to press the Reset button on the board to get it to connect), but clicking the little Sync Project button does nothing whatsoever.

# Steps to reproduce
This is starting from total scratch.

1. Open a new empty window in VS Code, click the "Create project" button under Pymakr's "Projects" section:
[image]
2. Add the serial port under the project:
[image]
3. Click the little lightning icon to connect and the spinner spins but never finishes:
[image]
4. If I click the "Create terminal" icon instead, I get a message `tty.usbmodem1234561 / unknown seems to be busy. Do you wish restart it in safe mode?` Clicking the "Restart in safe mode" button does nothing, but manually pressing the reset button on the board itself then causes the terminal window to connect and I'm now at the REPL which works:
[image]
5. Add tiny bit of code to `main.py` and click the "Sync project" button:
[image]

# Expected behavior
The project is uploaded and I can see the "Hello world" constantly being printed out every second in the terminal.

# Actual behaviour
Absolutely nothing happens, even when looking in the developer console with Pymakr's logs set to `debug` level.

# Environment
 - OS: macOS 12.4 on M1 Max MacBook Pro
 - Device: Unexpected Maker FeatherS2 (an ESP32S2-based board) running MicroPython 1.18.
 - VSCode: 1.67.2
 - Pymakr: 2.14.0 (preview)

# Additional context
I came across #242 and have updated the root path setting to be set to `/` for everything (I don't have any devices that don't have the root path at `/`):

```
    "pymakr.devices.config": [
    
        {
            "match": "manufacturer=.*",
            "field": "rootPath",
            "value": "/"
        }
    ],
```
But that hasn't made any difference unfortunately.

sinkers, 2022-06-08:
I am having pretty much exactly the same issue as this

jakobrosenberg, 2022-06-08:
Please let me know if this fixes your issue https://github.com/pycom/pymakr-vsc/issues/242#issuecomment-1140751462.

jakobrosenberg, 2022-06-08:
>  I'm happy to poke around and gather whatever info would be helpful for you

That would be much appreciated.

You can increase debugging level in settings:
[image]

Also, the Output tab  might contain more information
[image]

And in your settings, do you have an entry like this?

```json
    "pymakr.devices.configs": [
        
        {
            "autoConnect": "lastState",
            "name": "",
            "username": "micro",
            "password": "python",
            "hidden": false,
            "rootPath": "/flash",
            "id": "Py73bb20"
        },
        {
            "autoConnect": "lastState",
            "name": "",
            "username": "micro",
            "password": "python",
            "hidden": false,
            "rootPath": "/flash",
            "id": "Py3baa06"
        }
    ]
```

If so, could you please copy paste it to here.

VirtualWolf, 2022-06-08:
Ah yeah, the debug setting is something I stumbled onto in another issue, that's what I've had turned on with all those screenshots in my initial issue. 😄

[image]

I hadn't see the Output tab before, but it's displaying all the same stuff that the developer console does anyway.

As for the settings.json, yep, it's set like this:

```
    "pymakr.devices.config": [
        {
            "match": "manufacturer=.*",
            "field": "rootPath",
            "value": "/"
        }
    ],
```

If I change `pymakr.devices.config` add an `s` at the end like in your example, it just says it's an unknown setting.

jakobrosenberg, 2022-06-09:
Can you try Pymakr 2.15.x and see if that fixes it?

VirtualWolf, 2022-06-09:
_Slightly_ more progress with 2.15.0 in that I get this in the Output tab:

```
debug: PyMakr (26037) > Device: tty.usbmodem1234561 > busy...
```

But nothing else. Weirdly, when I clicked the Disconnect icon, at _that_ point I got a little VSCode dialog thing popping up saying it was uploading:

[image]

jakobrosenberg, 2022-06-09:
Did the update populate your `pymakr.devices.config`?

jakobrosenberg, 2022-06-09:
Sorry. I meant `pymakr.devices.configs`. `2.15.0` should create a new entry.

VirtualWolf, 2022-06-10:
Hm, no, it doesn't recognise `configs` with an `s` on the end. I deleted all of the `pymakr.*` entries from `settings.json` and uninstalled 2.15.0 entirely then reinstalled, and the default examples it recreates are under `pymakr.devices.config`, and if I change it to `configs` it's not recognised as an option:

[image]

rand12345, 2022-06-10:
Does anyone have an alternative extension or environment for MicroPython on M1? We have a project on hold here with no functional IDE.

No connection to devices, no terminal session to the devices.

The terminal process "node '/Users/xxx/.vscode/extensions/pycom.pymakr-preview-2.16.4/src/terminal/bin/client.js'" failed to launch (exit code: 1).

[image]
[image]
[image]
[image]

REPL is operational using screen, device is working just fine. No means in which to upload files, virtually no PyMakr options in VS Code. NodeJS is latest LTS and working.

This is some serious breaking change from older versions of PyMakr!

jakobrosenberg, 2022-06-10:
@rand12345 you can still install v1.

@VirtualWolf there was a bug in a dependency that didn't manifest itself in my local environment. I applied a fix, but I don't have any 3rd party devices to test it with. You can try it in `v2.16.4`.

VirtualWolf, 2022-06-11:
I'm afraid it's gone backwards now. 😂 I can't connect at all even after resetting the device, and when clicking the button to open the terminal window, I get `The terminal process failed to launch: Path to shell executable "node" does not exist.`

jakobrosenberg, 2022-06-13:
Thanks for the update @VirtualWolf .

What version are you currently on and would it be possible for us to have a remote session over Discord? My ID is `402846333087776768`

As for the the output panel. It just logs the data, it's not able to hijack it.

jakobrosenberg, 2022-06-14:
**UPDATE:** After a session with @VirtualWolf it seems that some 3rd party devices do not work with the `micropython-ctl` package. The REPL is not detected correctly, leaving the device in a busy state.

I'll provide a follow up when we have more info or a fix.

VirtualWolf, 2022-06-26:
We did some more testing and it seems like the root of the problem was the default chunk size and delay between sending chunks that the `micropython-ctl` library that Pymakr relies on uses. Pymakr 2.18.2 has two new settings you can configure in `pymakr.devices.configs` for your devices, and the following worked for me with an Unexpected Maker FeatherS2:

```json
"pymakr.devices.configs": [
  {
    "autoConnect": "onLostConnection",
    "name": "FeatherS2",
    "adapterOptions": {
        "chunkDelay": 20,
        "chunkSize": 256
    },
    "rootPath": "/",
    "id": "123456"
  }
]
```

Josverl, 2022-06-29:
do you know what are the defaults are?

jakobrosenberg, 2022-06-29:
If I remember correctly they're 3000 and 0 for size and delay.

VirtualWolf, 2022-06-29:
Yep: https://github.com/metachris/micropython-ctl/blob/master/src/main.ts#L725-L726

jakobrosenberg, 2022-07-20:
I assume this was fixed with the added `adapterOptions`. If not we can reopen it.

closed 2022-07-20 by jakobrosenberg

VirtualWolf, 2022-07-20:
Ah sorry. Yes indeed, everything's working a treat now, thanks!

Davocrack, 2024-01-02:
Hi, I have the same problem with an esp8266, and because I'm new in this world, I have no idea on where or how to put those codes. Can anyone help me? Mi discord id is davionga#1156

Josverl, 2024-01-02:
> Hi, I have the same problem with an esp8266, and because I'm new in this world, I have no idea on where or how to put those codes. Can anyone help me? My discord id is davionga#1156

best to create a new issue on this repo , if you want to save typing link to this one.