pycom/pymakr-vsc #260: [BUG] pymakr open in device explorer stops pylance from working

Josverl, 2022-06-27:
pymakr open in device explorer stops pylance from working when a MCU is attached and the rootpath of the device is '\'

(likely is similar but much smaller problem is also found for devices with `\flash` rootpath )
this was reported in : https://github.com/Josverl/micropython-stubs/issues/580
please refer to this for additional info 

**To Reproduce**
See : https://github.com/Josverl/stubs-esp-test/blob/main/repro.md

**Expected behavior**
Pylance to work as expected when a MCU is mounted in the explorer view 

**Desktop (please complete the following information):**
 - OS: Win 11 X64
 - Device : esp8266 
 - rootDrive : /
 - Pymakr : v2.18.2

jakobrosenberg, 2022-06-29:
I have considered exposing the full path, but the gains vs unforeseen implications make it a low priority. Is it correct that this is not a bug of Pymakr but an issue between VSCode and Pylance?

Josverl, 2022-06-29:
Indeed I can repro outside of Pymakr

closed 2022-06-29 by jakobrosenberg