pycom/pymakr-vsc #265: Pasting code with CRTL+V in terminal window does not work [BUG]

jandleuven, 2022-07-17:
I upgraded to Pymakr v2.12.3 in VS-Code 1.69.0.  Pymakr v2 is a great improvement over v1.  But pasting something on the clipboard  with CRTL+V into the terminal window seems not to work anymore in v2.

Luckily one can still use 'Edit' (from top menu bar of VS-Code), and next 'Paste'.  That menu also says that CTRL+V is the shortcut for it.  But that shortcut seems not to work in v2.12.3 of Pymakr.  Am I correct?

jakobrosenberg, 2022-07-18:
Hi @jandleuven . Thanks for the kind words and for reporting this. Could you try the [Preview version](https://marketplace.visualstudio.com/items?itemName=pycom.pymakr-preview) and tell me if this issue is present there.

jandleuven, 2022-07-18:
I disabled pymakr v2.12.3 and next installed the preview version v2.20.0.  But I still have the problem that CTRL+V does not work (while edit > paste from the menu works).
With v2.20.0 I can now upload code to the device, while that didn't work yesterday with v2.12.3.

jakobrosenberg, 2022-07-18:
I might have found the culprit. In the meantime, if you open a terminal from the menu here, it should work.

[image]

jakobrosenberg, 2022-07-18:
Hmm, this got weird fast. Opening the terminal from the devices / projects menu always works for me and also did just now. After opening it from terminal dropdown menu, I can no longer copy paste. I'll follow up as soon as I know more.

jakobrosenberg, 2022-08-15:
Closing this as VSCode seems to favor `ctrl + shift + v` by convention. 

If there's anything we can do about this at plugin level, I'll be happy to reopen the issue.

closed 2022-08-15 by jakobrosenberg