pycom/pymakr-vsc #266: [BUG]

kiwibrew, 2022-07-20:
**Describe the bug**
Visual Studio Code 1.69.2 (Universal)
on MacOS 12.4 (21F79)
with Pymakr 2.12.3

(and)
Visual Studio Code 1.69.2
on Ubuntu Linux 22.04
with latest Pymakr2

does not reliably communicate with LoPy4 on Pysense 1.1

Tested with three different LoPy4 + Pysense 1.1 device pairs in two different compute environments.

**To Reproduce**
1. Take a LoPy4 on Pysense
2. Update to latest firmware for both Pysense and LoPy4 (tried both app based and pybytes web based methods)
3. Open Visual Studio Code
4. Connect device to new project
5. Upload of LED example does not work

**Expected behavior**
When I press click the "sync project to device" icon, I expect the Led example main.py to upload. Instead nothing happens.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS 12.4 (21F79)
 - Device LoPy4 on Pysense 1.1
 - Pymakr 2.12.3

**Additional context**
Add any other context about the problem here.

jakobrosenberg, 2022-07-20:
Hi @kiwibrew, thanks for reporting this. Could you test if you also have this issue in our [upcoming version](https://marketplace.visualstudio.com/items?itemName=pycom.pymakr-preview)?

I'm hoping to release the preview on the stable channel at the end of this week/start of next.

kiwibrew, 2022-07-22:
The upcoming version is better able to communicate with the devices, thank you.

closed 2022-07-22 by kiwibrew