pycom/pymakr-vsc #270: Don't overwrite files when creating project

quintesse, 2022-08-02:
I pointed the plugin to a folder with existing files when clicking the "Create project" button and it overwrote `boot.py` and `main.py` I don't think it should do that.

Either there should be a button for "Import existing code" and refusing to create a project in a non-empty folder or the "Create project" button should just leave any existing files alone and only create those files if they don't exist.

The former is probably the most user-friendly while the latter is likely the easiest to implement.

jakobrosenberg, 2022-08-03:
Thanks for the feedback @quintesse . I think you're right.

We should check if the folder is empty and if it's not, prompt the user about overwriting existing files.

Something like
```
The folder isn't empty. How do you wish to handle existing files?
[Ignore them] [Overwrite them]
```

I expect I should be able to address this tomorrow or on Friday.

quintesse, 2022-08-03:
That certainly a possibility, but I would suggest to at least change the "Ignore them" to something more neutral like "Keep them". Because ignoring seems to me that the plugin won't do anything with them, so the user might get confused and think that neither of the two options is what they want. (For the user the distinction that the PyMakr plugin doesn't actually "do" anything with their files might not be very clear, so best use wording that is less confusing for them).

jakobrosenberg, 2022-08-04:
I agree. Removing "them" also improves readability. At least for me.

```
The folder isn't empty. How do you wish to handle existing files?
[Keep] [Overwrite]
```

jakobrosenberg, 2022-08-15:
This was added to the Preview release and will make it into the next stable build. Thanks again for the feedback @quintesse .

closed 2022-08-15 by jakobrosenberg