pycom/pymakr-vsc #271: uploadProject command fails with rmdir OSError: no such file or directory: 'null'

dajtxx, 2022-08-03:
I have installed pymakr in VSCode on MacOS. I can talk to the fipy via screen, or the terminal in VS code, started from the pymakr plugin.

When I try to upload the sample project I get this error:

[image]

pymakr.conf has this content:

```
{
    "name": "Led Example"
}
```

jakobrosenberg, 2022-08-03:
Hi @dajtxx . Could I ask you to post a screenshot of your device info? You can get it by hovering over your device.

<details>
<summary>Example</summary>

On windows VSCode can't be active when you hover over it or the overlay will disappear when you click win+shift+s (screenshot shortcut). I don't know if it behaves the same on Macs, but rather be safe than sorry. 🙂

[image]

</details>

dajtxx, 2022-08-08:
Hi @jakobrosenberg , here it is. I cannot connect to it to get better info.

[image]

jakobrosenberg, 2022-08-09:
Judging from the overlay, it seems Pymakr hasn't had write access to the terminal of your device. If you click `ctrl + f` in the terminal, does that give you a `>>>`?

Also, does the hover overlay give you the same info if you connect the device first?

dajtxx, 2022-08-10:
Here is what I see using a fipy that allows me to connect to the REPL:

[image]

You can see even with this one, that is connected, I can't get whatever details would show under Project or System. But I can talk to it via vscode terminal.

The bad one just never gets to the REPL.

I cannot get the pymakr vscode plugin to do anything much with these boards, other then get me to the REPL in the terminal.

I have now been able to download older firmware to the bad one and it's talking again. It is interesting this one with the oldest firmware I could get from the update tool shows more info in the vscode popup than the unchanged fipy.

[image]

dajtxx, 2022-08-10:
I still cannot get the upload project command to work reliably on macos, even though the dev has perms 0666. sudo code . makes no difference either.

It seems to be more reliable on Windows, but even then not completely.

I flashed back to a legacy firmware because I have no interest in pybytes and it seemed to want to connect to pybytes upon boot after a failed upload.

jakobrosenberg, 2022-08-10:
If you can't access to the REPL at all, you can try flashing the firmware with the `Erase during update` checked. If you still can't access the REPL after erasing the device, I suspect your device may be faulty.

[image]

For clarification, Pymakr can only communicate with your device through the REPL, so this has to be fixed before anything else.