pycom/pymakr-vsc #272: [BUG] Buttonas are not clickable / command pymakr.createProjectPromp not found

jairov96, 2022-08-04:
**Describe the bug**
Just installed pymakr today on both a windows device and a mac (m1 pro) device and realized all the buttons don't do anything.

When trying to use the commands through the "f1"  menu, I get this error. Same error both on mac and windows.

**To Reproduce**
Steps to reproduce the behavior:
1. Clock "f1"  in your keyboard
2. type "PyMakr: Create new project" 
3. Click on the first option
4. See error

**Expected behavior**
The prompt to create project to work

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: win10 / Mac Os 10 M1 pro
 - Device ESP32 wroom
 - Pymakr Latest available as of 04/08/2022

jakobrosenberg, 2022-08-04:
Thanks for reporting this @jairov96 . I'll try get a fix out ASAP.

jakobrosenberg, 2022-08-04:
I was unable to replicate this locally. Do you by any chance have both Pymakr and Pymakr Preview installed? This could mess up the install of either.

jairov96, 2022-08-04:
Hello @jakobrosenberg 

Turns out it's required to have Node.js installed for this to work and I wasn't aware of it. As soon as I installed, it's working like a charm. Huge thanks to everybody involved in this fantastic project.

Maybe would be worth pointing it out in the description that Node.js is indeed required to have this running.

Regards!

closed 2022-08-04 by jairov96

jakobrosenberg, 2022-08-07:
This is very interesting. Node should not be required for Pymakr.

Thanks so much for the kind words.

jairov96, 2022-08-07:
Just to point out the source where I got that:
https://randomnerdtutorials.com/micropython-esp32-esp8266-vs-code-pymakr/

At least, it solved my problems!