pycom/pymakr-vsc #275: [BUG] "machine" changed to "fake_machine"

rob-baker, 2022-08-14:
In "development mode" when I save a file and it is automatically uploaded to the device, the following line is changed:

VS Code editor:
`import machine, onewire, ds18x20, time`

Changed on the device to:
`import fake_machine, onewire, ds18x20, time`

Why is the "machine" changed to "fake_machine"?

Right clicking on the file in VS Code and selecting "pymakr > Upload to device" works ok.

Dev setup:
VS Code Version: 1.70.1 (system setup)
Commit: 6d9b74a70ca9c7733b29f0456fd8195364076dda
Date: 2022-08-10T06:08:33.642Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Windows_NT x64 10.0.19044
Pymakr: v2.22.4

jakobrosenberg, 2022-08-15:
Thanks for reporting this Rob. Could you check if this is still an issue in `v2.22.5`?

jakobrosenberg, 2022-08-15:
> Why is the "machine" changed to "fake_machine"?

This was caused by a missing check for `pymakr.conf -> dev.simulateDeepSleep`. It's still WIP and should not have made it into your build.

rob-baker, 2022-08-15:
That fixed it. Thank you.

I'm new to pymakr and love it!

jakobrosenberg, 2022-08-15:
Thanks so much @rob-baker .

I'll close this issue as resolved.

closed 2022-08-15 by jakobrosenberg