pycom/pymakr-vsc #276: cannot click button 'create project'

AlanaAckermans, 2022-08-19:
Type: <b>Bug</b>

cannot click button 'create project', i have tried uninstalling and installing both vscode and the extension but it stays the same issue. i also cannot open a workspace or do anything at all

Extension version: 2.22.5
VS Code version: Code 1.70.2 (e4503b30fc78200f846c62cf8091b76ff5547662, 2022-08-16T05:35:13.448Z)
OS version: Windows_NT x64 10.0.22000
Modes:

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-1035G1 CPU @ 1.00GHz (8 x 1190)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: disabled_off<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>opengl: enabled_on<br>rasterization: enabled<br>raw_draw: disabled_off_ok<br>skia_renderer: enabled_on<br>video_decode: enabled<br>video_encode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.60GB (1.73GB free)|
|Process Argv|--crash-reporter-id 08228522-154e-4aef-b5db-5e01a954cecf|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383:30185418
vspor879:30202332
vspor708:30202333
vspor363:30204092
vslsvsres303:30308271
pythonvspyl392:30443607
vserr242:30382549
pythontb:30283811
vsjup518:30340749
pythonptprofiler:30281270
vshan820:30294714
vstes263:30335439
vscorecescf:30445987
pythondataviewer:30285071
vscod805:30301674
binariesv615:30325510
bridge0708:30335490
bridge0723:30353136
cmake_vspar411cf:30542925
vsaa593:30376534
pythonvs932:30410667
wslgetstarted:30449410
cppdebug:30492333
pylanb8912:30545647
vsclangdf:30486550
c4g48928:30535728
d7fjc500:30543455
dsvsc012:30540252
azure-dev_surveyone:30548225

```

</details>

jakobrosenberg, 2022-08-19:
Hi @AlanaAckermans . Thanks for reporting this. Do you have this issue with both `Pymakr` and `Pymakr - Preview`? Both are available in the VSCode store.

If so, can I ask you to post your logs from VSCode? You can open them by clicking `Help` -> `Toggle Developer Tools`.

AlanaAckermans, 2022-08-19:
Hi @jakobrosenberg, I indeed have this issue with both pymakr extensions. These are the logs from VSCode :

[Extension Host] --- Starting Pymakr ---
console.ts:137 [Extension Host] 12:05:17.486 info: > [Pymakr] > pymakr v2.22.5
log.ts:313   ERR [Extension Host] 'node' is not recognized as an internal or external command,
operable program or batch file.
console.ts:137 [Extension Host] 'node' is not recognized as an internal or external command,
operable program or batch file.
I @ console.ts:137
mainThreadExtensionService.ts:111 Activating extension 'Pycom.pymakr' failed: Command failed: node -v
'node' is not recognized as an internal or external command,
operable program or batch file.
.
$onExtensionActivationError @ mainThreadExtensionService.ts:111
localProcessExtensionHost.ts:298 Extension Host
localProcessExtensionHost.ts:299 'node' is not recognized as an internal or external command,operable program or batch file.
log.ts:313   ERR command 'pymakr.createProjectPrompt' not found: Error: command 'pymakr.createProjectPrompt' not found
    at p._tryExecuteCommand (vscode-file://vscode-app/c:/Users/alana/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:1692:3532)
    at p.executeCommand (vscode-file://vscode-app/c:/Users/alana/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:1692:3414)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async i.open (vscode-file://vscode-app/c:/Users/alana/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:602:47316)
    at async l.open (vscode-file://vscode-app/c:/Users/alana/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:602:49092)
3log.ts:313   ERR command 'pymakr.createProjectPrompt' not found: Error: command 'pymakr.createProjectPrompt' not found
    at p._tryExecuteCommand (vscode-file://vscode-app/c:/Users/alana/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:1692:3532)
    at p.executeCommand (vscode-file://vscode-app/c:/Users/alana/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:1692:3414)
    at async i.open (vscode-file://vscode-app/c:/Users/alana/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:602:47316)
    at async l.open (vscode-file://vscode-app/c:/Users/alana/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:602:49092)
contextKeyService.ts:452 Element already has context attribute: pane, expanded, horizontal

G234i, 2022-09-06:
Similar issue here.
Installing Node.js (v18.8.0) seems to fix the issue and "Create project" works.
@AlanaAckermans try installing a version of node.js and see if your problem persists?

@jakobrosenberg if node.ds is a dependency maybe it should be listed in GET_STARTED.md and/or README.md?

jakobrosenberg, 2022-09-07:
@G234i it used to be a dependency, but it shouldn't be anymore.

G234i, 2022-09-07:
> @G234i it used to be a dependency, but it shouldn't be anymore.
> 
> EDIT: there's a mention here https://github.com/pycom/pymakr-vsc/blob/next/FAQ.md#do-i-need-node

A quick look at [AlanaAckermans](https://github.com/AlanaAckermans)'s logs - something is calling "node -v", and indeed [src/PyMakr.js](https://github.com/pycom/pymakr-vsc/blob/18d04db020da90066d43de62886777626ee7a86a/src/PyMakr.js) contains that on line 111.

So I guess that there are still some leftover calls made towards node.js in the code.

PS: I also believe that I saw some "npm" calls in the code and that would be another node.js dependency, unless it is installed standalone by the pymakr extension?

jakobrosenberg, 2022-09-07:
@G234i well spotted. That line is actually supposed to detect if node is present and if not, then use prebuilt binaries. It seems it throws an error when node isn't detected.

I've published a new fix on the preview-branch. If it works I'll push it to the stable branch.

I can't test it locally, but it's available on v2.25.2 .

G234i, 2022-09-08:
I can confirm that with v2.25.2 the project creation process works after uninstalling node.js.
PS: It also works with node.js aswell.

jakobrosenberg, 2022-09-08:
Thanks @G234i I'm delighted to hear that. And thanks again for catching spotting the culprit. 😊

jakobrosenberg, 2022-09-08:
I'm closing this as resolved as the fix is not available in the preview version.

closed 2022-09-08 by jakobrosenberg