pycom/pymakr-vsc #277: M5Stack ATOM with MPY v1.19.1 not capable of syncing files

Wenlin88, 2022-09-11:
What I did: Running Sync project to device,

What I expected to happen: Files transferred from project to device

What actually happened:  Extensions freezes to uploading prompt

Additional info:  As seen below locationId is not defined. From windows device manager you can find that COM port properties have Location: On USB serial converter. Can this be reason why ATOM is not working with Pymakr?

[image]

|Config||
|-|-|
|autoConnect|onLostConnection|
|name||
|username|***|
|password|***|
|hidden|false|
|rootPath|/|
|adapterOptions|{}|

|Device||
|-|-|
|path|COM9|
|manufacturer|FTDI|
|serialNumber|6552F13692|
|pnpId|FTDIBUS\VID_0403+PID_6001+6552F13692A\0000|
|locationId|undefined|
|friendlyName|USB Serial Port (COM9)|
|vendorId|0403|
|productId|6001|

|System||
|-|-|
|memFree|109488|
|sysname|esp32|
|nodename|esp32|
|release|1.19.1|
|version|v1.19.1 on 2022-06-18|
|machine|M5Stack ATOM with ESP32-PICO-D4|
|uniqueId|d8a01d5957ec|
|fsBlockSize|4096|
|fsBlocksTotal|512|
|fsBlocksFree|510|

|Host||
|-|-|
|OS|win32 - x64|
|Pymakr|undefined|
|Pymakr-Preview|2.25.2|
|VSCode|1.71.0|

## Device History at 12:11:02.528

|function|args|time|result|queued at|finished at|failed|
|-|-|-|-|-|-|-|
|getFile|<details> <pre>[<br />  "null/pymakr.conf"<br />]</pre> </details>|never finished|<details> <pre>null</pre> </details>|11:57:52.185|null|false|
|listFiles|<details> <pre>[<br />  "/"<br />]</pre> </details>|159|<details> <pre>[]</pre> </details>|11:57:52.250|11:57:52.185|false|
|getBoardInfo| |142|<details> <pre>{<br />  "memFree": 109488,<br />  "sysname": "esp32",<br />  "nodename": "esp32",<br />  "release": "1.19.1",<br />  "version": "v1.19.1 on 2022-06-18",<br />  "machine": "M5Stack ATOM with ESP32-PICO-D4",<br />  "uniqueId": "d8a01d5957ec",<br />  "fsBlockSize": 4096,<br />  "fsBlocksTotal": 512,<br />  "fsBlocksFree": 510<br />}</pre> </details>|11:57:51.882|11:57:52.250|false|

jakobrosenberg, 2022-09-16:
Hi @Wenlin88, sorry about the slow response.

It looks like your device is never initialized properly. Could you check your Pymakr configuration and let me know what the rootPath of your device is set to? You can find it under `settings -> Pymakr -> devices:Configs`.

It seems to be `/`, but for some reason Pymakr is looking at `null`.

Wenlin88, 2022-09-19:
Hi @jakobrosenberg, Thank for your answer. This is how my device config looks like;

"6552F13692": {
  "autoConnect": "onLostConnection",
  "name": "ATOM",
  "username": "micro",
  "password": "python",
  "hidden": false,
  "rootPath": "/",
  "adapterOptions": {}
}