pycom/pymakr-vsc #280: Pymakr cant add project folder

alphagreen84, 2022-10-15:
Hey, 
I have tried both pymakr plugins in VS CODE but none is working.

On the preview version i got the following error when trying to add a project folder:

[Pymakr] Failed to run command: createProjectPrompt. Reason: Cannot read properties of undefined (reading 'replace'). Please see logs for info.
Where do I find the logs?
Any idea what is wrong?

- OS: Ubuntu 22.04 LTS
- Pymakr v2.25.2
- VS CODE 1.72.2
**Additional context**

calston, 2023-03-23:
Same problem. Seems to work if you install the Preview version

xiaohe4966, 2024-10-08:
> Same problem. Seems to work if you install the Preview version

Thank you, I really can

SilverNodashi, 2026-02-21:
I just ran into the same problem, running:

- Linux Mint 21.34
- Visual Studio Code Version 1.109.2
- NodeJS 24.13.1

`[Pymakr] Failed to run command: createProjectPrompt. Reason: Cannot read properties of undefined (reading 'replace'). Please see logs for info.`