pycom/pymakr-vsc #289: [BUG] Upload prompt won't finish

Leo5126, 2022-12-13:
**Describe the bug**
When trying to upload a file, the upload prompt will open and it also starts uploading. The problem is, that the window won't close, and the communication with the device is also blocked. To have access on the device again a restart of VSC is required. The interesting thing is, altough it doesn't look like the upload worked, the file is uploaded correctly when i recheck it after a restart. 
Is this a known issue?

**To Reproduce**
in my case this error occurs every time i try to upload a file by right clicking on file and then upload to device.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
[image]

**Desktop:**

- OS: [Win 10 x64]
- Device [ESP32S3]
- Pymakr [2.22.5]

daniel-da, 2022-12-14:
Hi Leo,

It may be a known issue, can you try the fix I've made in this PR?

https://github.com/jakobrosenberg/micropython-ctl/pull/2

Leo5126, 2022-12-15:
Hi @daniel-da,

Thanks for the tip. But I don't quite understand how to test this other repo together with the extension. 
How did you do this?

daniel-da, 2022-12-16:
Hi @Leo5126 ,

First you get the file content from the PR. 
This is the direct link: 
  https://raw.githubusercontent.com/jakobrosenberg/micropython-ctl/40121187cd12b242eab2eb7c36c196c9a87b4064/src/main.ts

Then copy it into the extention folder, I'm on linux so it is here for me: 
  /home/dany/.vscode/extensions/pycom.pymakr-2.22.5/node_modules/micropython-ctl-cont/src

Hope that will fix you issue

Leo5126, 2022-12-21:
Hi daniel,

Unfortunately this doesn't work for me. I get a new error when trying to upload:

[image]

You know this error? 

I've changed the file for the preview version of pymakr plugin 2.22.5

hsand, 2023-01-22:
I'm experiencing similar issues on Ubuntu with the ESP32S3 as well. I tried the pull request @daniel-da mentioned above, didn't get any errors as @Leo5126 did, but it also didn't fix the issue.

I'm wondering if this is somehow related to the S3 variant of the ESP32.

sausthapit, 2023-02-01:
> Hi daniel,
> 
> Unfortunately this doesn't work for me. I get a new error when trying to upload:
> 
> [image]
> 
> You know this error? Do I need to deactivate any auto update or smth similar?
> 
> I've changed the file for the preview version of pymakr plugin 2.22.5

Hi new to Pycom, the upload doesn't work for me either. I am using MacOs and LoPy4 with Pysense. Same error as shown in the picture. Did you manage to solve it?

miguelcostaUI, 2023-03-21:
I also can't upload any code. It just says that it is uploading forever and nothing happens.
Tried the normal version (v2.22.5) and the preview version (v2.25.2) and nothing is uploaded.
macOS 12.6.3 - VS Code Version: 1.76.2

Did you find any way to fix this?