pycom/pymakr-vsc #293: [BUG] Commands missing in VS Code

Diapolo10, 2023-02-02:
**Describe the bug**
I've been trying to work with MicroPython on an ESP32 Feather (HUZZAH32), following this tutorial: https://randomnerdtutorials.com/micropython-esp32-esp8266-vs-code-pymakr/

However, it seems my installation of Pymakr is missing functionality. There are no entries on the bottom bar (probably because other extensions have used up most of it), which is fine, but the command menu is also devoid of most of them: https://i.imgur.com/0seuDPr.png

I can supposedly connect to the device via the left-hand panel, and with that I can open a MicroPython REPL, but it doesn't seem like uploading is doing anything and some packages such as `pycom` are missing/unavailable.

I would've uploaded this issue from within the extension; but there was no debug menu option anywhere.

**To Reproduce**
I'm not certain if the problem is reproducible, but here are the steps I took:

1. Install VS Code (I already had this installed (list of extensions: https://pastebin.com/jbDCcmZ2)
2. Install Pymakr
3. Plug in ESP32 Feather
4. Install MicroPython using esptool
5. Connect to ESP32 via VS Code
6. Write a Hello World program
7. Attempt to upload (nothing happens)

**Expected behavior**
The program should run on the ESP32 and print Hello World via the serial port

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- OS: Windows 10 x64 22H2 Build 19045.2546
- Device: ESP32 Feather (V1, presumably)
- Pymakr v2.22.5

**Additional context**
I'm using a GPD Pocket 3, in case that matters.

faanskit, 2023-02-16:
I face a similar problem. With a new install of v2.22.5  "There are no entries on the bottom bar".
* MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
* VSCode Version: 1.75.1 (user setup)
* Date: 2023-02-08T21:32:34.589Z
* Electron: 19.1.9
* Chromium: 102.0.5005.194
* Node.js: 16.14.2
* V8: 10.2.154.23-electron.0
* OS: Windows_NT x64 10.0.19045
* Sandboxed: No

I can however download _Sync Project to Device_  **works OK** as well as use the terminal.
But, when I try to **Run Active File** I get:
```
>>> c:\Users\User\Documents\Micropython\test\main.py
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```
As the author, I have a limited set of commands:
[image]

As the author, I miss the bottom bar - and in particular - a way to Run
Screenshot:
[image]

mhavill, 2023-10-12:
Same issue with Pymakr Console and missing commands in the palette:  

    OS: Win 10 X64
    Device ESP32 board
    Pymakr v2.25.2

However, I am working around these issues and able to download/upload manage ESP32 - THANK YOU