pycom/pymakr-vsc #296: [BUG] Cannot start development mode

mkelk, 2023-03-21:
**Describe the bug**
When using "Pymakr: Start development mode .....", I get an error from VS Code saying

[Pymakr] Failed to run command: startDevMode. Reason: Cannot destructure property 'project' of 'undefined' as it is undefined.. Please see logs for info.

I have also created a new Pymakr project from within VS Code, using "Pymakr: Create new project", just to make sure that it was not the project set-up that caused the error.

**Expected behavior**
I was expecting development mode to activate.

**Desktop (please complete the following information):**

- OS: Win 10 X64
- Device ESP32 board
- Pymakr v2.12.5

mhavill, 2023-10-12:
I also have the same problem.  Where do I find the 'Logs'

Desktop (please complete the following information):

    OS: Win 10 X64
    Device ESP32 board
    Pymakr v2.25.2 (preview)

psy0rz, 2023-12-17:
same problem on ubuntu 22

Josverl, 2024-01-28:
the company pycom does not exist any more.
there is no maintenance anymore. You could see if [MicroPico](https://marketplace.visualstudio.com/items?itemName=paulober.pico-w-go) may be a relevant option.
it is based off pymakr , and allows for Pico and other devices.

mhavill, 2024-01-28:
Thank yo for the response