pycom/pymakr-vsc #300: Pymakr projectRef not set

dh-digid, 2023-06-02:
What I did: Installed Pymakr extension on VS-Code, successful connected to device on COM8, able to access device, upload and download files.
Trying to use CTRL+SHIFT+S to store files easier to the device

What I expected to happen: The file is stored correctly

What actually happened: [Pymakr] Failed to run command: uploadPrompt. Reason: projectRef did not match an accepted type. Please see logs for info.

Additional info: When hoovering over the device it tells me Project name undefined. Maybe this is the problem, but there is no option or configuration to set the project name.
How can I setup the correct host project name?

|Config||
|-|-|
|autoConnect|always|
|username|***|
|password|***|
|hidden|false|
|rootPath|/|
|adapterOptions|{}|
|name|ESP32_Azure|
|projectName|ESP-VSC|

|Device||
|-|-|
|path|COM8|
|manufacturer|Silicon Labs|
|serialNumber|0001|
|pnpId|USB\VID_10C4&PID_EA60\0001|
|locationId|Port_#0004.Hub_#0006|
|friendlyName|Silicon Labs CP210x USB to UART Bridge (COM8)|
|vendorId|10C4|
|productId|EA60|

|System||
|-|-|
|memFree|93696|
|sysname|esp32|
|nodename|esp32|
|release|1.20.0|
|version|v1.20.0 on 2023-04-26|
|machine|ESP32 module with ESP32|
|uniqueId|30aea4dc8394|
|fsBlockSize|4096|
|fsBlocksTotal|512|
|fsBlocksFree|490|

|Host||
|-|-|
|OS|win32 - x64|
|Pymakr|undefined|
|Pymakr-Preview|2.25.2|
|VSCode|1.78.2|

## Device History at 13:15:14.678

|function|args|time|result|queued at|finished at|failed|
|-|-|-|-|-|-|-|
|getFile|<details> <pre>[<br />  "//pymakr.conf"<br />]</pre> </details>|never finished|<details> <pre>null</pre> </details>|13:14:46.340|null|false|
|getBoardInfo| |158|<details> <pre>{<br />  "memFree": 93696,<br />  "sysname": "esp32",<br />  "nodename": "esp32",<br />  "release": "1.20.0",<br />  "version": "v1.20.0 on 2023-04-26",<br />  "machine": "ESP32 module with ESP32",<br />  "uniqueId": "30aea4dc8394",<br />  "fsBlockSize": 4096,<br />  "fsBlocksTotal": 512,<br />  "fsBlocksFree": 490<br />}</pre> </details>|13:14:46.181|13:14:46.340|false|
|getFile|<details> <pre>[<br />  "//pymakr.conf"<br />]</pre> </details>|never finished|<details> <pre>null</pre> </details>|13:14:46.600|null|false|
|getBoardInfo| |162|<details> <pre>{<br />  "memFree": 93696,<br />  "sysname": "esp32",<br />  "nodename": "esp32",<br />  "release": "1.20.0",<br />  "version": "v1.20.0 on 2023-04-26",<br />  "machine": "ESP32 module with ESP32",<br />  "uniqueId": "30aea4dc8394",<br />  "fsBlockSize": 4096,<br />  "fsBlocksTotal": 512,<br />  "fsBlocksFree": 490<br />}</pre> </details>|13:14:45.898|13:14:46.600|false|

shnlmn, 2023-10-01:
I have the exact same issue. I'm running VSCode on Windows 10.

benoitLecuppe, 2023-11-09:
metoo