pycom/pymakr-vsc #305: Failed uploadProject command due to rmdir OSError: -2

riffnshred, 2023-09-23:
What I did: Sync my project to it. Note that I use the pimoroni-pico version of Micropython for raspberry pi pico

What I expected to happen: To get my code on to it

What actually happened: Files did not sync and VScode showed this error. 

[Pymakr] Failed to run command: uploadProject. Reason: Traceback (most recent call last): File "<stdin>", line 20, in <module> File "<stdin>", line 6, in rmdir OSError: -2. Please see logs for info.

Additional info: I'm working on a project that use a Pimoroni Display Pack 2.0 with a raspberry pi pico, so I'm using their version of Mycropython, Pimoroni-pico. This setup used to work until now when I created a new project.

|Config||
|-|-|
|autoConnect|null|
|name||
|username|***|
|password|***|
|hidden|false|
|rootPath|null|
|adapterOptions|{}|

|Device||
|-|-|
|path|COM6|
|manufacturer|Microsoft|
|serialNumber|e6614103e76d3b23|
|pnpId|USB\VID_2E8A&PID_0005&MI_00\7&2EC08CB3&1&0000|
|locationId|0003.0000.0000.012.003.001.000.000.000|
|friendlyName|USB Serial Device (COM6)|
|vendorId|2E8A|
|productId|0005|

|System||
|-|-|

|Host||
|-|-|
|OS|win32 - x64|
|Pymakr|undefined|
|Pymakr-Preview|2.25.2|
|VSCode|1.82.2|

## Device History at 07:00:29.636

|function|args|time|result|queued at|finished at|failed|
|-|-|-|-|-|-|-|
|remove|<details> <pre>[<br />  null,<br />  true<br />]</pre> </details>|never finished|<details> <pre>null</pre> </details>|06:56:52.360|null|false|
|getBoardInfo| |never finished|<details> <pre>null</pre> </details>|06:55:47.200|null|false|

Josverl, 2025-11-24:
The company pycom does not exist any more, and therefore there is no maintenance to pymakr anymore. 
You could see if [MicroPico](https://marketplace.visualstudio.com/items?itemName=paulober.pico-w-go) may be a relevant option.
It is based off Pymakr , and allows for Pico and other devices.

alejanpa17, 2025-12-10:
Hello @minaolenpavel, the development mode worked for me as a workaround :)

[image]