pycom/pymakr-vsc #311: pymakr not working

dhruvesh2907, 2024-01-14:
Type: <b>Bug</b>

Error running command pymakr.createProjectPrompt: command 'pymakr.createProjectPrompt' not found. This is likely caused by the extension that contributes pymakr.createProjectPrompt.

Extension version: 2.22.5
VS Code version: Code 1.85.1 (0ee08df0cf4527e40edc9aa28f4b5bd38bbff2b2, 2023-12-13T09:49:37.021Z)
OS version: Windows_NT x64 10.0.22631
Modes:

carpweb, 2024-01-15:
Same here.

VSC:
Version: 1.85.1 (user setup)
Commit: 0ee08df0cf4527e40edc9aa28f4b5bd38bbff2b2
Date: 2023-12-13T09:49:37.021Z
Electron: 25.9.7
ElectronBuildId: 25551756
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22631

Josverl, 2024-01-28:
the company pycom does not exist any more.
there is no maintenance anymore. You could see if [MicroPico](https://marketplace.visualstudio.com/items?itemName=paulober.pico-w-go) may be a relevant option.
it is based off pymakr , and allows for Pico and other devices.