pycom/pymakr-vsc #312: Why can't I create a project with this error?:

sadsagads, 2024-02-26:
**IMPORTANT**

If possible, please consider creating the issue from within Pymakr. This will create a new issue template with relevant context information.

To do so, click the device's context menu (`...`) -> `Debug` -> `Show Device Summary` -> `Create an issue on Github`

If you prefer to use this form instead, you can delete the IMPORTANT section.

---

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- OS: [e.g. Win 11 X64]
- Device [e.g. Expansionboard v3.1]
- Pymakr [e.g. v2.12.3]

**Additional context**
Add any other context about the problem here.

romain145, 2024-04-25:
Adding this in the [Getting Started](https://github.com/sg-wireless/pymakr-vsc/blob/HEAD/GET_STARTED.md#getting-started) would make sense.

[image]

JEELJEEL, 2025-01-29:
I experienced the same issue.

Thank you very much @peck-lack for the solution.