pycom/pymakr-vsc #316: [BUG] Cannot Upload to Connnected Device

drew-wks, 2024-10-15:
**Describe the bug**
Identical to the issues described in #[253](https://github.com/sg-wireless/pymakr-vsc/issues/253) 
I can connect to the device, but cannot upload/download, soft reset or stop script on the device.

This is starting from total scratch.

Open a new empty window in VS Code, click the "Create project" button under Pymakr's "Projects" section:
1_new_project
[image]

2. Create a templated project
[image]

3. Add the serial port under the project:
[image]

4. Click the little lightning icon to connect
Console confirms connection with `[Extension Host] 11:38:35.549 info: > [Pymakr] > Device: tty.usbserial-1D52E20475 > Connected.`
[image]

6. Device appears to be accessible:
Console reports: 
`[Extension Host] 11:38:35.549 info: > [Pymakr] > Device: tty.usbserial-1D52E20475 > Waiting for access...`
`Extension Host] 11:38:35.565 info: > [Pymakr] > Device: tty.usbserial-1D52E20475 > Got access.`
`[Extension Host] 11:38:35.565 info: > [Pymakr] > Device: tty.usbserial-1D52E20475 > Getting device info.`
[image]

 
7.  and the spinner spins but never finishes:
[image]

8. If I click the "Create terminal" icon instead, it connects and I'm now at the REPL which works:
```
MicroPython v1.23.0 on 2024-06-02; M5Stack ATOM with ESP32-PICO-D4
Type "help()" for more information.
>>> 
```

9. Add tiny bit of code to `main.py` and click the "Sync project" button

Expected behavior
The project is uploaded and I can see the "colored LEDs on the device.

Actual behavior
Nothing happens, even when looking in the developer console with Pymakr's logs set to debug level.

Device Info
M5Stack ATOM with ESP32-PICO-D4
MicroPython v1.23.0 (2024-06-02)

Dev Environment
Pymakr 2 v2.22.5
Node.js v20.11.0
VS Code 1.94.2 (Universal)
Macbook Air M2, 2022 running Sonoma 14.1

Additional context
I came across https://github.com/sg-wireless/pymakr-vsc/issues/243, but that hasn't made any difference unfortunately.

```
  "pymakr.devices.configs": {
    "1D52E20475": {
      "autoConnect": "never",
      "name": "ATOM Lite",
      "username": "micro",
      "password": "python",
      "hidden": false,
      "rootPath": "/",
      "adapterOptions": {
        "chunkDelay": 20,
        "chunkSize": 256
      },
      "fast_upload": false,
      "baudrate": 115200
    },
```

During Step 2 when loading pymakr, the debug logs show:
`ERR No registered selector for ID: ms-vscode.npm-command`

During Step 3 when loading pymakr, the debug logs show the following. Neiter "getting device info" nor "upload" receive a confirmation.
```
[Extension Host] 12:04:19.353 info: > [Pymakr] > Device: tty.usbserial-1D52E20475 > Connected.
console.ts:137 [Extension Host] 12:04:19.353 info: > [Pymakr] > Device: tty.usbserial-1D52E20475 > Waiting for access...
console.ts:137 [Extension Host] 12:04:19.377 info: > [Pymakr] > Device: tty.usbserial-1D52E20475 > Got access.
console.ts:137 [Extension Host] 12:04:19.377 info: > [Pymakr] > Device: tty.usbserial-1D52E20475 > Getting device info.
console.ts:137 [Extension Host] 12:04:31.136 info: > [Pymakr] > Device: tty.usbserial-1D52E20475 > upload /Users/drew_wilkins/Drews_Files/Drew/Python/bug_report to /
```

jaketeater, 2024-11-27:
I had a similar issue on my generic ESP32-WROOM. I was able to connect, but not upload/download/soft reset/stop script.

I found that if I connect right away and immediately clicked on stop script, it would work. But this only worked about 1 out of 5-10 tries.

I had this problem in both VS Code (PyMakr) and Thonny. 

I had flashed (and reflashed) with MicroPython v1.24.0 (ESP32_GENERIC-20241025-v1.24.0.bin)

I was able to resolve the issue by using ESP32_GENERIC-20240602-v1.23.0.bin

I realize this isn't going to help you, since you are already using 1.23.0, but I am commenting in case someone else would find it useful.

Josverl, 2024-11-27:
The company pycom does not exist any more, and therefore there is no maintenance to pymakr anymore. 
You could see if [MicroPico](https://marketplace.visualstudio.com/items?itemName=paulober.pico-w-go) may be a relevant option.
It is indirectly based on Pymakr , and allows for Pico and other devices.